jankae/LibreVNA #129: Add tile(s) to VNA display

nbgsmk, 2022-07-08:
### Summary

How to add tiles to the VNA display?
For example, one graph is displayed on the main screen of the VNA.
If a user wants to split the display, in order to add another tile, for another graph, there is no way to do it. User has to close the existing graph first, then split the tile. By doing this, the setup of the initial graph is lost, and it can be time consuming to set it up from the beginning.
Actually, this goes for any number of graphs displayed: how to add another tile, without closing an existing tile first?

### Paths to implement it

Add option to right-click the existing graph and 'add tile' or 'split horz/vert' or similar. This seems the most logical action.
Alternatively, add an option to the main menu, under 'window' or something like that.

### Extra information

n/a

jankae, 2022-07-09:
Great idea, this has been annoying me for a while as well. With 80eaf5b you get an additional option to split a tile while keeping the already configured graph:
[image]

nbgsmk, 2022-07-09:
Truly great :-)  and I admire your speed.
One question, though: I see this section-title "Traces" but in my platform it is not visible. It is another suggestion I was aiming for, since a long time ago. So...any idea why it is missing in my case? 

[image]

jankae, 2022-07-09:
The "Traces" section is just a named separator in the context menu:
[image]
It might not be supported on all platforms, in that case you'll just get a normal separator line as in your screenshot. You can not click on "[Primary/Secondary] Traces", you are not missing any functionality.

nbgsmk, 2022-07-09:
I understand. I was not aiming to click on it, but the title primary/secondary is very helpful to me. I find myself always pausing for a moment, thinking why i see the trace list twice. :)
If that's a platform-dependent issue, well, what to do... you are welcome to close this issue.
Thanks!

jankae, 2022-07-09:
At least I think it is platform-dependent, see also here: https://doc.qt.io/qt-5/qmenu.html#addSection

> "The rendering of the hint is style and platform dependent. Widget styles can use the text information in the rendering for sections, or can choose to ignore it and render sections like simple separators."

Sounds to me like it might be possible to somehow use a different style for the rendering and show the name? Not sure...

> you are welcome to close this issue.

Will do :)

closed 2022-07-09 by jankae