jankae/LibreVNA #130: Command line options

nbgsmk, 2022-07-08:
### Summary

By loading (admittedly) a _very_ old .setup file, I managed to garble my gui and cause an application crash on every subsequent start (windows platform).
Only by deleting the .setup file in question, I was able to get back into the default configuration and continue from there.

Therefore a general question: are there any command line options for the pc application? 

Specifically, an option to ignore setup, calibration etc, and load everything default, would be great.

### Paths to implement it

Three variants, starting with simplest:
- just a simple command line option to load all defaults (ie. ignore any remembered .setup from before)
- command line option to load a specific .setup file. If used with empty value, then load all defaults.
- alternative: command line option for a specific .setup file, another command line option for 'defaults'

### Extra information

I can provide logs during app startup in this problematic situation, together with the .setup which causes the crash. But I believe it is not worth troubleshooting, because it is my own fault for loading a quite old .setup file. Please let me know.

jankae, 2022-07-09:
First of all, I would be interested in the .setup file which managed to crash the GUI. While very old .setup files may not completely work as intended (some features might have been added/removed or renamed), they should never cause a crash.

There are a couple of command line options:
[image]
As you can see, you are able to specify a setup file, but there is no way of **not** loading the default setup. And even if you specify a different setup file here, the default setup file is loaded first and then the GUI is reconfigured with the specified file.

nbgsmk, 2022-07-09:
Okay, here some more details:
First of all, I am using this option
[image]

This gives me the most complete 'last used' state, including traces and markers, as I tend to fiddle with a single topic for days.
How to cause the 'crash'?
Load the setup file below (from maybe 1.2 or earlier release, honestly not sure). I don't really remember what is inside of it, so wanted to open it, check it, and probably delete afterwards.  But Myrphy's law never sleeps. :)
[2 panel 3mkr_HAM.setup..zip](https://github.com/jankae/LibreVNA/files/9076611/2.panel.3mkr_HAM.setup.zip)

After loading, the gui looks like this. It is not really messed up so badly but, as you see, there is only 'device log' available, and under toolbars, only 'reference' is shown.

[image]

[image]

After that, if you exit the app and try to start it again, I believe this is happening: 
a.setup has gotten corrupted by autosaving on exit (here is a.setup at this point  
[a.setup.zip](https://github.com/jankae/LibreVNA/files/9077252/a.setup.zip)
so attempts to start the app again will fail everytime.
If you delete a.setup, the app will just gracefully skip it and create a new one which is ok.

Finally, here is the app startup sequence during the crash. Nothing interesting, just crashes at some final stage.
[app start sequence and crash on corrupted setup file.txt](https://github.com/jankae/LibreVNA/files/9077265/app.start.sequence.and.crash.on.corrupted.setup.file.txt)

At this point, I found a related issue just now: if the a.setup is completely empty (0 bytes) it will also crash the app during start. The message is 

 0.812: [warning] Parsing of setup file failed:  [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal

Therefore I see two things here:
- setup file with unreliable content, so to speak. In my case "2 panel 3mkr_HAM.setup". It will cause some kind of problem later in the app, which can not be detected during startup phase. For this I propose a --no-setup or --default-session or a similar kind of command line option. Alternative is to delete the offending .setup file, but what if you can't be sure which one it is? Or something in the saved preferences crashes the app?

- setup file obviously corrupted, like in the case of empty content, which can be detected during app start (like the above "parsing of setup file failed"). This could be handled with either a command line option from the first point (need user action), or by gracefully skipping the .setup parsing and loading default config instead. 

I already found how to add command line options, but I would welcome your comments first.

jankae, 2022-07-09:
> After loading [2 panel 3mkr_HAM.setup], the gui looks like this.

Can confirm, I got the same problem. This is caused by not activating any mode. Your setup file was so old that it didn't contain the active mode (VNA was always active after loading a setup file). Without any active mode, most toolbars/docks are not available. I have fixed the setup loading code to activate the first mode found if none is specified. This setup file now loads correctly.

> I believe this is happening:
a.setup has gotten corrupted by autosaving on exit (here is a.setup at this point [...] so attempts to start the app again will fail everytime.

That is correct, the GUI saved the current state into a.setup and because no mode was active, a.setup now also contains no indication on which mode to start. The same fix also takes care of this. 

> At this point, I found a related issue just now: if the a.setup is completely empty (0 bytes) it will also crash the app during start.

Yes, I forgot to abort parsing after catching the first exception. Also fixed now. Invalid json in the setup file will no simply abort loading the setup.

> For this I propose a --no-setup or --default-session or a similar kind of command line option.

The problem I see is to define the default. Depending on your preferences this might be a number of things. Or maybe just have a command option to start with default preferences? This only requires calling Preferences::getInstance().setDefault() instead of Preferences::getInstance().load() during startup. As everything else is determined by the preferences, the GUI also wouldn't load any setup file with the default preferences.

nbgsmk, 2022-07-09:
Confirmed, works great.
Both with a 'corrupted' or an empty setup file, the app loads normally. Json parsing error shows a warning dialog, so the user is aware of the situation. That's really user friendly.

> Your setup file was so old that it didn't contain the active mode (VNA was always active after loading a setup file). Without any active mode, most toolbars/docks are not available.

I was suspecting that adding a new mode/tab manually in the top right may improve the situation, but the goal is not to be forced into finding a workaround. 

> Or maybe just have a command option to start with default preferences? This only requires calling Preferences::getInstance().setDefault() instead of Preferences::getInstance().load() during startup.

If you are thinking this
[image]

then yes. 
Exactly that was my idea. 
Developer's own defaults should be the safest of all.

jankae, 2022-07-10:
> If you are thinking this

Not just this but all preferences, everything that can be changed via window->preferences. This will definitely get you the default settings as they were when you first launched the application. I have added this as a command line option now (--reset-preferences)

nbgsmk, 2022-07-10:
That's even better!

closed 2022-07-10 by nbgsmk