jankae/LibreVNA #131: Delta marker reference to...

nbgsmk, 2022-07-08:
### LibreVNA Version

LibreVNA Version (64 bit): 1.4.0-ade970b35
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

- create a few markers
- delete some of them from the 'middle' of the list, creating non-consecutive indexes
- now the marker dock is populated with markers for example 1, 2, 6, 7
- change one of them to 'delta' marker (eg. 7)
- try to change that 7 is delta to 6 by mouse clicking on up/down arrows
- the reference will cycle in this sequence: 0 (invalid delta), 1, 2,  0, 1 2...etc (never reaches 6)
- fortunately it is possible to type 6 manually

[image]

### Expected behavior

Delta marker reference cycles through available indexes: 1, 2, 6, 7, even if they are not consecutive. 

Ideally 1, 2, 6 (as there is no reason to be delta to itself).
But 0, 1, 2, 6, 7 is also acceptable. 
At least for my personal taste. :-)

### Extra information & Setup and Calibration files

n/a

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

jankae, 2022-07-09:
Yes, using a QSpinBox for the delta selection doesn't make a lot of sense if you can have "holes" in the marker numbers (which happens when you delete markers). I have now changed it to a QComboBox which only contains the valid delta marker numbers (displayed in the same order as in the marker dock):
[image]
In this example, marker 3 is on a time domain trace which makes it unusable as delta marker for marker 2.

nbgsmk, 2022-07-09:
That's even better than I expected. :-)

closed 2022-07-09 by nbgsmk