jankae/LibreVNA #132: Smith Chart Displays

YuryW, 2022-07-09:
### LibreVNA Version

[image]

### Steps to reproduce

The LibreVNA application is running. The LibreVNA device is not connected.

### Expected behavior

Not all input parameters can be loaded into Smith charts for s3p and s4p files, in this case the S22 imported file. Or am I doing something wrong?
[image]
[image]
And can you make it so that the extensions of imported third-party sXp files would be case-insensitive, since S3P (S4P) files often come across and the program now swears at them?

### Extra information & Setup and Calibration files

[TCM1-83X+_S3P.zip](https://github.com/jankae/LibreVNA/files/9077694/TCM1-83X%2B_S3P.zip)

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [X] Touchstone file of measurements

jankae, 2022-07-09:
For now, you can only add reflection measurements to a smith chart. In your example, S22 **is** a reflection measurement and should work.
However, there was an error in the code that determines whether a trace is a reflection measurement when the imported touchstone file has more than two ports. That is why you can instead add S21, which should not be possible.
I just fixed this (3af9b73), it now works as expected.

jankae, 2022-07-09:
> And can you make it so that the extensions of imported third-party sXp files would be case-insensitive, since S3P (S4P) files often come across and the program now swears at them?

Almost forgot about that one. Importing .S3P is now also possible without an error message.

YuryW, 2022-07-09:
Simply amazing! So quickly fixed! Best regards.
We close the question.

jankae, 2022-07-09:
Happy to help :)

closed 2022-07-09 by jankae