jankae/LibreVNA #134: Limit lines import/export

nbgsmk, 2022-07-11:
### Summary

Hello @jankae 
do you consider it would be usefull to add import and export of limit lines to the axis setup dialog? 
At the moment, they are only saved/restored in the overall setup file. 
Because it involves a bit of manual typing, it could be helpful to export them for reuse in another setup. What do you yhink?

### Paths to implement it

Add import/export buttons here and connect them to already existing toJSON / fromJSON methods.
I already have some of the code and I could complete it relatively easy.
[image]

### Extra information

Also considered importing just the points of a single line (in the edit points dialog) but this seems more elegant.

jankae, 2022-07-11:
Yes, seems like a great feature with a reasonable simple implementation. If you already started on that, feel free to create a pull request once it is ready :)

nbgsmk, 2022-07-11:
Sure, I will let you know.

nbgsmk, 2022-07-11:
PR #135 sent. Closing this proposal, let's continue there.

closed 2022-07-11 by nbgsmk

YuryW, 2022-07-15:
Greetings dear friends. I wanted to ask for a new feature to control and customize the touchstone viewer window. And is it possible that you are already trying to implement this function? I understand correctly that we are talking about imposing on a rectangular graph (with the ability to turn off the mode with a checkmark) additional line segments, the number of which and their color can be adjusted, and the coordinates of the ends of the segments can be set in pairs (dB / Hz - dB / Hz), so that they are displayed in the program window on rectangular measurement graphs and serve to control and (or) configure devices according to their current measurements or downloaded sXp graphs?

nbgsmk, 2022-07-16:
Hello! 
If you are thinking this, then yes, it is already available, thanks to Jan.
As you see, it is also possible to choose colours and check for pass/fail decision.
I am not sure if it was already included in the latest release 1.3.0, which you can download directly. But if you have the environment to build the executable yourself, then in 1.4.0 this is how it looks. I hope, this answers your question?

[image]

YuryW, 2022-07-16:
Thank you dear friend for your detailed answer. I will look forward to the collected next release 1.4. Yes, this is almost 100% what I wanted.  
Tell me please, is it planned to implement the calculation of unevenness or is it already in version 1.4?
That is, between two markers, which are determined during the settings in the menu, are calculated and displayed: 
1) the gain value in dB at the point of the marker with a lower frequency, 
2) the decrease or increase (slope) in dB to the second marker, 
3) the frequency difference between the markers , 
4) as well as (most importantly, the deviation) the maximum in "+" and the minimum in "-" deviation in dB from the straight line connecting both markers. True, I still did not understand through which point the line passes in VNA Planar, relative to which everything is calculated.
 Somewhere near the second marker. This feature is called Irregularity.

nbgsmk, 2022-07-16:
Hi! Thanks for your interest! About the first 3 points, you can create one marker manually and another as delta marker. Then differences between them will be displayed, eg. like delta freq. and delta dB. At least what I understand from your question. 
About 4, I am not sure I completely understand, but maybe @jankae can comment if it already exists or is in plan. This nice instrument well exceeds my work as it is, so there are quite a few features which I didn't even try to explore. :)
Another thing, if you have specific questions, bugs or issues which are outside of this scope, I kindly encourage you to open new issues. Otherwise, intermediate topics may get lost in a thread like this. 
For general questions or discussions, the [LibreVNA group](https://groups.io/g/LibreVNA-support/) is also a good place.

YuryW, 2022-07-17:
Hello [nbgsmk](https://github.com/nbgsmk).
Thanks for answers. Of course, then I will try to ask questions so that they are not lost in correspondence on other occasions. Already got a lot for myself in the LibreVNA group.
Best regards.

jankae, 2022-07-17:
Can confirm, 1-3 should be achievable with a delta marker. Number 4 is not really possible at the moment (although you might be able to set something like this up with the custom math expressions, but it wouldn't be user friendly). I also think that this would be quite a specific feature to implement, you might want to look into the SCPI API and do that calculation yourself on the S parameter data.

YuryW, 2022-07-17:
Hi! Here is the exact description. 
[image]

This method of measuring gain variation (or flatness) is often used for amplifiers with an equalizer function (such as cable TV amplifiers). The method is quite simple - the difference in dB between the transient response (Sxy) and the line segment between two markers at all measurement points and selecting the maximum difference in dB above and below the line segment and adding them. It differs from the simple delta calculation of the markers in that it takes into account the relative change in transient response with respect to the straight line segment between them.
This measurement is important in many cases - in the passbands of various filters, wideband and narrowband amplifiers. Sorry that I am writing here again - so that it does not turn out that I am opening a new request (after all, I first asked here).

> you might want to look into the SCPI API

It is unlikely. )) I'm looking forward to the official release of LibreVNA 1.4, although many LibreVNA fans have already built it themselves...
I promise another question-request for the function to be done soon in Issues. :))
It will also be very important for many, a little more difficult to implement, but it will raise the level of interest in LibreVNA by an order of magnitude compared to competitors...