jankae/LibreVNA #136: Imported Touchstone traces do not follow span

nbgsmk, 2022-07-12:
### LibreVNA Version

LibreVNA Version (64 bit): 1.4.0-97fceb4ff
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

After importing a trace from touchstone file and adding it to a graph, changing start, center, stop, span scales the graph in an unusual way.
Example file contains measurements from 10MHz to 1.1GHz.   
[Direkciona sprega FWD, kabl_SUHNER.zip](https://github.com/jankae/LibreVNA/files/9097496/Direkciona.sprega.FWD.kabl_SUHNER.zip)

After adding a trace from s2p to a primary axis, the graph spans what seems to be a wider range of the two: imported s2p  (red) or current span of the live trace (yellow).  
Is this the expected behaviour?
[image]

### Expected behavior

Graph always spans according to the values in the acquisition toolbar.
The idea is to capture wideband s2p measurements with many points, and later be able to recall and zoom in for comparison with a live trace in great detail.

### Extra information & Setup and Calibration files

n/a

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [X] Touchstone file of measurements

jankae, 2022-07-13:
Depends on your graph settings, both options are available:
[image]
Fit span: Graph shows the frequency range selected in the toolbar (should match your expected behavior)
Fit traces: Graph shows complete traces (looks like this option is enabled in your screenshot)

nbgsmk, 2022-07-14:
Oh! 
I never noticed it until now. 
Thanks, it works great!