jankae/LibreVNA #138: GUI: touchstone viewer

horror-vacui, 2022-07-14:
### Summary

It would be a nice and I hope easy to implement feature to display one and two port touchstone files (.s1p, .s2p). So instead of reading out the data from the LibreVNA device, they would be loaded from a file.

An additional nice option would be if such imported data traces could be permanently added to the display, while a live readout (i.e. continuous measurement) from the LibreVNA is happening.

I am a new user, and though I have been searching for this option I have not found it. Let me know if something similar has already been implemented.

### Paths to implement it

Replace data coming from the device with data coming from a file.

### Extra information

.

jankae, 2022-07-14:
Already possible, check out this button:
[image]

horror-vacui, 2022-07-14:
> Already possible, check out this button

Many Thanks! I've only looked in the menus and into the "+" button, but missed that icon. I have a black system theme (and probably a terrible monitor), and lines in the import/export icons stayed black, making the drawing on the icon very hard to see.  The lines in the other icons where white, but not these ones. There is a similar issue in the traces field:
[image]
Please take it as information, and not a critic. I like the tool. Actually I am really glad that the GUI recognizes the system theme at all.

It would be great to load in multiple files, and not just a single one. I guess it is a limitation in the File API.

I could not copy the exact build from the About window (could be a nice feature it implement), but I use this version:
[image]
I am using it on Manjaro Linux with KDE. I could  imagine that most these little nuances are only present in my config.

Many thanks for the great software and hardware! Keep up the good work!

jankae, 2022-07-14:
Hm, I haven't tried it with a dark theme yet, probably quite a few things that don't look great. The problem is that a lot of the icons are just simple .pngs regardless of the theme. Most general GUI stuff is handled by Qt which seems to recognize the theme. But any custom icon will just stay exactly the same. I am not too familiar with GUI programming, especially when it comes to creating graphics. I might look into it at some point, but it probably won't happen soon, sorry.

> I could not copy the exact build from the About window (could be a nice feature it implement)

Should be possible:
[image]

> Many thanks for the great software and hardware! Keep up the good work!

Thank you, I'll to my best :)

closed 2022-07-17 by jankae