jankae/LibreVNA #141: Delayed reaction of the Smith chart pop-up submenu to changes in the reference resistance of this chart.

YuryW, 2022-07-28:
### LibreVNA Version

LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

### Steps to reproduce

LibreVNA application launched.
LibreVNA device is not connected.

### Expected behavior

The Smith chart should always display only touchstone files with the same reference impedances.

We have 2 touchstone files Test_R_75.s2p and Test_R_50.s2p, differing only in reference resistance. In Test_R_75.s2p it is 75 ohms:
[image] and in Test_R_50.s2p it is 50 Ohm:
[image].
Import both files into LibreVNA v1.4.0. 
[image]
The floating submenu on the Smith chart contains only those files whose reference impedance matches the reference impedance of the Smith chart. The default is 50 ohms.
You can change the Smith chart reference impedance in the Setup settings of this pop-up submenu, only in this case by unchecking the S11 LibreVNA characteristic:
[image]
Let's do this and set the reference impedance to 75 ohms:
[image]
We click OK and (surprisingly) all the same, in the pop-up submenu it is proposed to display touchstone files with a reference resistance of 50 Ohms, and not 75 Ohms, on the Smith chart.
[image]
Check the box next to Test_R_50_S11 (with a reference impedance of 50 ohms) and this file is displayed on the Smith chart with a reference impedance of 75 ohms. And only after that (after adding the first file to the Smith chart) in the pop-up submenu, files with a reference resistance of 75 Ohm are offered as needed:
[image]
Put a tick in front of Test_R_75_S11 and get:
[image]
The result on the Smith chart with a reference resistance of 75 ohms, files with a reference resistance of 50 ohms and 75 ohms were loaded at the same time, which, probably, was not expected:
[image]
As a result, this situation is possible due to the fact that after changing the reference impedance in the Setup settings of the pop-up submenu, before the first transfer of the touchstone file to the Smith Chart, the pop-up submenu offers to load the previous version with the previous reference impedance.

 

 

### Extra information & Setup and Calibration files

[Test_R_75.s2p_Test_R_50.s2p.zip](https://github.com/jankae/LibreVNA/files/9209136/Test_R_75.s2p_Test_R_50.s2p.zip)

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [X] Touchstone file of measurements

jankae, 2022-07-28:
Thank you for the excellent bug report. The context menu (pop-up) wasn't updated properly when changing settings of the smith chart (e.g. the reference impedance). 6a27d39 should fix that.

YuryW, 2022-07-28:
Thanks for the quick response and fixes as always. We close the question.
Best regards!

closed 2022-07-28 by jankae