jankae/LibreVNA #143: The LibreVNA application works in W7_64, but VNA itself does not.

YuryW, 2022-08-06:
### LibreVNA Version

LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

### Steps to reproduce

The LibreVNA device is connected to the computer. The system sees it as VNA. The LibreVNA application does not see the LibreVNA device.

### Expected behavior

The LibreVNA device is connected to the computer via USB-2.0. The Ready and Power LEDs glow green. Supply voltage 5.18 V, current consumption 0.58 A. The W7_64 system sees it as a VNA. Then the LibreVNA application turns on and an error window pops up. 
[image]
EmbeddedFirmware-hw-rev-B-v1.4.0 is installed in the device on W10_64.
In W10_64 everything is fine.

### Extra information & Setup and Calibration files

Designation Rev. B_6L means the board is 6 layer?
[image]

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

jankae, 2022-08-06:
Easy question first:

> Designation Rev. B_6L means the board is 6 layer?

Yes, that is the "new" version of the LibreVNA. "New" because not much has changed. Just some minor layout adjustments which lowered the noise floor a bit above 3GHz and also improved return loss slightly. No huge improvements and electrically identical (apart from some passives).

> The LibreVNA application does not see the LibreVNA device.

I don't have any machine running W7 anymore, so unfortunately I can't test this. But it could be a missing driver. The LibreVNA uses a USB descriptor which should tell windows what driver to load. Maybe that is not working properly on W7?
You can always try to assign the driver yourself: Use [zadig](https://zadig.akeo.ie/) and select the WinUSB driver for the LibreVNA.

YuryW, 2022-08-06:
Thank you very much!!! Zadig helped. The application began to work with the LibreVNA device in W7_64 as well. True, W7_64 now does not recognize LibreVNA as VNA, but this is not important. ))
Now I have VNA itself, so I’ll ask questions here more than once over time ...
And we close the issue as resolved. I hope that it will help not only me.

closed 2022-08-06 by jankae

YuryW, 2022-08-06:
And there will be no version for W7-32?

jankae, 2022-08-06:
It should be possible to compile the project for 32bit systems. Probably just a matter of setting up the github actions to do that as well. I didn't expect anyone still using 32bit, so I didn't bother so far.

YuryW, 2022-08-06:
> Я не ожидал, что кто-то все еще использует 32-битную версию, поэтому пока не беспокоился.

You (or me) can ask this as a separate question so that it can be seen by everyone who enters the questions. And to get an opinion, such as conduct a survey, how many users still use this system. It's just that I have one workplace for general work and there is a specific place for devices via USB on the old computer and it's still difficult to change it to a new one...))