jankae/LibreVNA #146: Smith chart frequency span

TopQuark12, 2022-08-08:
### Summary

Introduce min and max frequency range setting for plotting Smith charts in addition to current "Limit to current span" setting. 

This is useful when performing time gated measurements where data acquisition span is maximised to the device limit for better TDR resolution, yet the user does not wish to plot the whole frequency range on the Smith chart for the final gated data. 

This is not an issue for X-Y plots as the X axis range can be manually adjusted.

### Paths to implement it

Either add a setting to the Smith chart setup dialog to allow for manual control of frequency range of plotted data. 

Alternatively allow custom expression to conditionally assign a null value for data point with frequency outside of band of interest.

### Extra information

The time gate function and custom expressions is a hidden gem, keep up the good work!

jankae, 2022-08-09:
Sounds useful and also fairly easy to implement. You can expect this addition in about 2-3 weeks if no unexpected problems appear

jankae, 2022-08-21:
With 7123984, the smith chart (and the newly introduced polar chart) has a "visible frequency override setting":
[image]

closed 2022-09-04 by jankae