jankae/LibreVNA #147: Error, starting headless without display

spel-oe, 2022-08-12:
### LibreVNA Version

LibreVNA-GUI v1.4.0

### Steps to reproduce

ssh to device: 
./LibreVNA-GUI --no-gui -p 8080
     0.038: [warning] could not connect to display 
     0.038: [info] Could not load the Qt platform plugin "xcb" in "" even though it was found.
     0.039: [fatal] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted

### Expected behavior

with --no-gui parameter I would expect the software to start even without gui :)

### Extra information & Setup and Calibration files

-

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

sophiekovalevsky, 2022-08-12:
Even though you hit `./LibreVNA-GUI --no-gui -p 8080`, the LibreVNA gui interface will keep running on the background. 

Have you try to pass `-X` when doing ssh into it?

spel-oe, 2022-08-12:
starting ssh with -X is not an option since the LibreVNA software should run automated in the background with no interaction (GPIB/SCPI driven)

jankae, 2022-08-17:
Does your system have a display? If so, please try "export DISPLAY=:0" before starting the LibreVNA-GUI when connecting via ssh.
If your system is headless, you might be in trouble...

spel-oe, 2022-08-17:
sadly my pc (small, arm based, battery powered) has no display.  "Error, starting headless without display #147"
I thought that "--no-gui" is for such cases, that the program does not try to start anything related to x-server. 

For me it would be enough if I can state "no x-server" before I compile the PC-Application

jankae, 2022-08-17:
In theory, it should be possible to split the GUI and SCPI parts of the application. Sadly, the SCPI addition wasn't planned in the beginning and the code is very intertwined. It would be a lot of work to refactor all of it to work completely without a GUI. As mentioned earlier, the `--no-gui` option simply hides all the dialogs/windows but the code is still there running in the background.
I'd recommend a Raspberry Pi with the standard OS with x-server and just not connecting any display.

spel-oe, 2022-08-17:
reminds me to the nanovna-saver where I experienced the same, luckily I was able to fork and solve it by my own. Of course with losing the upstream updates.  

Thank you for both your answers,  so it would be a nice feature for the future :)

jankae, 2022-09-04:
I'll go ahead and close this without providing a satisfying solution, sorry. The task of separating GUI and logic seems just too daunting at this stage. The project could (likely should) have been planned like this in the beginning, but adding scripting functionality wasn't on my mind then.

closed 2022-09-04 by jankae