jankae/LibreVNA #148: Inconsistency of the numbers of the numbering of marker measurements with the labels themselves (triangles).

YuryW, 2022-08-25:
### Summary

LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 7 Version 6.1 (Build 7601)
CPU Arch: x86_64

Marker numbers on charts do not fit in their triangular labels. Is it possible to make, like other VNAs, the marker number above the triangular mark?
[image]

### Paths to implement it

And it would also be chic, probably, if the user could determine the position of the number in relation to the marker icon (triangle) himself in the menu bar of markers (lower right corner of the application) to the right of the data - numbers above (ab), below (be), to the left (le ) or to the right (ri) of the marker label.

[image]

### Extra information

An example of placing the marker number above the marker label in one of the VNAs.
[image]

jankae, 2022-09-04:
Thank you for the suggestion, the marker number are indeed very poorly readable. I have added a selection of different marker styles in the preferences, here is an example of how the markers can look now:
[image]

I did not add on option to display the number anywhere else than above the triangle. My opinion is that this is sufficient for almost all use cases and adding this option would introduce complexity to the software: the marker class generates the symbol, but its placement on the graph is done by the individual graph classes. With the number beside the triangle, the position at which the overall marker icon (number + triangle is drawn) would have to be adjusted accordingly, requiring additional information exchange between these classes. This is certainly doable and I am happy to accept PRs for this if anymore deems this feature absolutely necessary.

YuryW, 2022-09-04:
jankae, thanks a lot. A perfectly acceptable solution. I think that this question can be closed, so as not to waste more time. New issues will be addressed as they arise. I will wait for the new release.
Best regards.

closed 2022-09-04 by jankae