jankae/LibreVNA #151: Using large fonts in graphs on High DPI screens clips labels

korken89, 2022-10-08:
### LibreVNA Version

LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Arch Linux
CPU Arch: x86_64

### Steps to reproduce

Use a computer with 4k screen, set the font size for axises to 30.

[image]

### Expected behavior

That the labels do not clip.

### Extra information & Setup and Calibration files

N/A

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

jankae, 2022-10-08:
I do not have a 4k screen and using a font size of 30 with my monitor only allows me to partially reproduce the problem:
[image]
The X axis tick values are still clipped on the sides, but no values are clipped on the bottom. The clipping on the sides is due to a fixed maximum width for each label. I have replaced the fixed size with an adjustable one that depends on the font size, this fixed the clipping on the sides.

As I am unable to reproduce clipping on the bottom, this is probably still present in your setup. I checked the code for plotting the Y-axis label and it already scales the maximum box height according to the selected font size. Not sure what is going wrong with the high DPI monitor...

korken89, 2022-10-08:
Thanks for checking!

If you could point me to where in the code the scaling is, I could experiment a bit and maybe solve the remaining issue. :)

jankae, 2022-10-08:
Sure, you need to look in Software/PC_Application/LibreVNA-GUI/Traces/tracexyplot.cpp

Line 424 and 426 for the Y-axis tick labels, line 630-631 for the X-axis tick labels. This all assumes you are using the latest commit on master.

YuryW, 2022-10-08:
LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64
4K UHD 3840x2160
Set font size for axes to 50.
[image]

YuryW, 2022-10-21:
LibreVNA Version (64 bit): 1.5.0-alpha.1-c7a99af82
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64
Good evening! A little information on the recent version and on this issue. Monitor screen resolution 3840x2160.
The maximum font sizes for axes are 44 with markers and 54 without markers. Everything is normal, only there are frequency verticals, under which there are no frequency marks. This, with large fonts, introduces the user into difficulty a little.
Perhaps the frequency verticals with inscriptions can be made a little longer at the bottom.
[image]
[image]

jankae, 2022-10-21:
>  Everything is normal, only there are frequency verticals, under which there are no frequency marks.

Tick labels on the X axis are skipped if they would not fit (when ticks are too close to each other for the selected font size).

> Perhaps the frequency verticals with inscriptions can be made a little longer at the bottom.

Good idea :) Maybe the axis ticks outside of the plot area should scale with the label font size? At the moment they are fixed and only using two pixels. Looks good on my screen but in your screenshot with the higher resolution they are barely visible.

YuryW, 2022-10-21:
> Maybe the axis ticks outside of the plot area should scale with the label font size?

Yes, that would probably be the best solution.
And further. Is it possible to reduce the distance between Magnitude and Phase and the graph? So that there is more space for the graph due to the voids between Magnitude
 and graph on the left and Phase and graph on the right?

jankae, 2022-10-21:
>  Is it possible to reduce the distance between Magnitude and Phase and the graph?

You can reduce the font size, it also influences the size reserved for the Y axis tick labels. I don't really want to make it any closer, it would result in collisions betweeen "Magnitude/Phase" in certain cases (depending on the actual values and unit of the Y axis). But maybe an option could be added in the preferences to manually set it to your liking...

YuryW, 2022-10-21:
> But maybe an option could be added in the preferences to manually set it to your liking...

No, don't. Perhaps, when the most important thing in LibreVNA is implemented in accordance with your ideas, then it will be possible to return to such trifles...))

closed 2023-10-23 by jankae