jankae/LibreVNA #152: Calibration measurements dialogue box

Chorbu, 2022-10-13:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.1-4ae7b31a4
OS: Windows 11 21H2 Version 22000.1098
CPU Arch: x86_64

### Steps to reproduce

On the dialog box CALIBRATION MEASUREMENTS when i try to add OPEN or SHORT or others standards, the application crashes, LibreVNA GUI suddenly closed without any error dialogue box. It seems it appear only on Windows version (on two differents PCs with Win11. On LINUX UBUNTU it doesn't crash with the same version of LibreVNA GUI.

Thank you so much,
Patrick F5SUE

### Expected behavior

Bug correction

### Extra information & Setup and Calibration files

No extra informations

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

jankae, 2022-10-13:
Interesting, I can replicate that when I download the compiled version from Github but when I compile on my machine and execute that, there is no crash. That will make debugging a bit more difficult...

jankae, 2022-10-13:
Please check the result from this run: https://github.com/jankae/LibreVNA/actions/runs/3246096610
I just removed some warnings in the compiler run and didn't actually expect to fix anything by that but this version works for me on Windows 10

Chorbu, 2022-10-14:
Hello Jan,

It works well now ! Thank you so much. Have a good day. Patrick

Chorbu, 2022-10-14:
Jan,

There is still a small problem to fix. Yes the SOLT (full two ports) calibration is OK now. But when i click ACTIVATE button the LibreVNA-GUI crashes like as i said on the top of this issue.

So i can't activate any calibration. Thank you Patrick

jankae, 2022-10-14:
Hi Patrick,
I can't be sure but my best guess is that you don't have any calibration standards defined. The selection in the "Calkit Standard" column must not be empty:
[image]
If you can't select anything, you must create the calibration standards in the calibration kit first. Of course, you can save all these to files so you don't have to go through the whole process every time.

Clicking activate without any calibration standards resulted in an attempt to access a non-existing calibration standard which crashes the application. I have now changed the behavior and you can't activate a calibration anymore unless all the calibration standards are defined.

jankae, 2022-10-14:
> Yes it works now.

Awesome :)

> Sorry for forgetting creating CAL standard before.

No need to be sorry, thank you for discovering the bugs. I'll look into the default cal name as well.

> Have you some informations about development of LibreCAL ?

Hardware and software design is done, I am currently waiting for the first prototype.

Chorbu, 2022-10-14:
OK thank you Jan. I'm sure LibreCAL will be very nice to use and simpler ! I usually use it in my work with agilent VNA's about 18 Ghz. Sure i will buy one !! Thank you. Patrick

jankae, 2022-10-16:
I'll go ahead and close this as the original issue seems to be solved. Feel free to open a new issue if you find any other problems or bugs :)

closed 2022-10-16 by jankae