jankae/LibreVNA #154: Application LibreVNA Version (64 bit): 1.5.0-alpha.1-c7a99af82 expands to full screen on the third try.

YuryW, 2022-10-21:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.1-c7a99af82
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

### Steps to reproduce

Only the LibreVNA application is running. The device is not connected.
After launching the application, it is first located in the upper left corner. After clicking the mouse on a full screen scan, it moves a little lower and to the right. And only after the next mouse click on the full screen scan, everything turns out.
[image]
[image]
[image]

### Expected behavior

For version: 
LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64
The application immediately expanded to full screen.
Perhaps I missed something?

### Extra information & Setup and Calibration files

No.

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

YuryW, 2022-10-21:
Tried Default values, Last used, Setup file. All the same. Okay, it's not that hard to double click the mouse. Perhaps, over time, it will be decided or no one will use W7. ))

YuryW, 2022-10-24:
LibreVNA Version (64 bit): 1.5.0-alpha.1-796a8cb3f
OS: Windows 10 21H2 Version 19044.1288
CPU Arch: x86_64
For Windows 10 and with a screen resolution of 3840x2160, the same pattern repeats as in the first question.

For all subsequent commits, up to today, when you try to run the LibreVNA_GUI application on Win7 or Win10, the following splash screen appears.

LibreVNA-GUI.exe - System error. 
 Unable to continue code execution because the system has not detected libusb-1.0.dll . To fix this problem, try reinstalling the program.

dna35, 2022-12-11:
I noticed the same 3 time click operation but have ignored it until seeing this post.

LibreVNA Version (64 bit): 1.5.0-alpha.2-b67275831
OS: Windows 10 Version 2009
CPU Arch: x86_64

Display #2 or far left of three displays sometimes goes off screen on the right side and can't see the bottom parts of the GUI unless not in max window size. I'm using DualMonitor tools but unknown if it is related. However, don't know when it will do it yet.

jankae, 2022-12-11:
If I can replicate this myself I would attempt to fix it. But I haven't ever seen this, so I am unlikely to find the problem.
I'll go ahead and close this for now. If you (or anyone else) has new information about this, I'll happily reopen this.

closed 2022-12-11 by jankae