jankae/LibreVNA #155: Axis Setup->X axis Power - Default max is 6 GdBm. Trimming the level label in the lower left corner on the left and right.

YuryW, 2022-10-22:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.1-796a8cb3f
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

### Steps to reproduce

LbbreVNA application is running. The device is not connected.
Trace Display Area ->XY-plot->Axis Setup->X axis Power - Default max is 6 GdBm. Trimming the level label in the lower left corner on the left and right.
[image]

### Expected behavior

Probably the default X-axis should be Minimum -40 dBm, Maximum 0 dBm. Or something close to the characteristics of the device.

### Extra information & Setup and Calibration files

No.

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

jankae, 2022-10-25:
Okay, several small things in this issue:

> Trace Display Area ->XY-plot->Axis Setup->X axis Power - Default max is 6 GdBm

- While in the axis setup dialog, no autoranging is performed. The selected settings are only applied once you close the dialog. This means that you will still see the old min/max values when switching from one X axis type to the other, even when autoranging is active
- When the x axis is displaying power, autoranging only supports the "fit traces" mode
- In your example there are no traces (or at least no traces with datapoints), so autoranging doesn't know what to do and just leaves the minimum/maximum as it was
- If you now add a trace, autoranging should work as intended and adjust your X axis settings

I have now improved this slightly. In case autoranging doesn't find any traces, it uses default min/max values instead of just leaving them as they are. The default min/max values depend on the X axis type (frequency/power). If a device is connected, its maximum/minimum supported values are used as the limits. If no device is connected, some hardcoded default values are used instead.

> Trimming the level label in the lower left corner on the left and right.

This bug was recently introduced when fixing some display issues for larger font sizes. Should be fixed now.

> Axis setup -> X axis : after switching from Frequency (Power) to Time or Distance, you cannot switch back to Freqency (Power) without exiting this (Axis Setup) menu and re-entering.

This one I can't replicate. I can switch between any X axis settings for any number of times without closing the dialog inbetween.

YuryW, 2022-10-25:
> In case autoranging doesn't find any traces, it uses default min/max values instead of just leaving them as they are.

Now in LibreVNA Version (64 bit): 1.5.0-alpha.1-2dd4edfae
a little wrong. When switching X axis to Power with no device connected, the X range is set to -100dBm ... 30dBm. And in Sweep type Power the default is -30dBm...-10dBm. That is, these settings are not synchronized?
[image]

jankae, 2022-10-25:
> That is, these settings are not synchronized?

No, they are not. They would be for autorange mode "use span", but I haven't implemented that yet for the X axis in power mode.

What you are getting are the default values when no device is connected.

YuryW, 2022-10-25:
> This one I can't replicate.
We switch in X axis from Frequency to Time, then again to Frequency. Here's what I have with the charts...
The frequency graph will already be different.
[image]
[image]
[image]
[image]

YuryW, 2022-10-25:
> > That is, these settings are not synchronized?
> 
> No, they are not. They would be for autorange mode "use span", but I haven't implemented that yet for the X axis in power mode.
> 
> What you are getting are the default values when no device is connected.

Thanks Jankae. It's clear.

jankae, 2022-10-25:
> We switch in X axis from Frequency to Time, then again to Frequency. Here's what I have with the charts...
The frequency graph will already be different.

The switch from frequency to time disables the Y axes, because the previously selected axis types are not available in the time domain. When you switch back to frequency, they are still disabled (the software doesn't store the previously selected type). If you select magnitude (for the primary Y axis) and phase (for the secondary Y axis), you should get the original frequency graph back.

YuryW, 2022-10-25:
It's clear. And here too, Jankey. Maybe when switching Sweep Frequency to Sweep Power, or change graphics to other ones automatically, or not change the frequency on the remaining graphs?
[image]
[image]

jankae, 2022-10-25:
Ah, I haven't tested that specific case. The graphs can switch the axis type automatically, if the trace data changes. It only works halfway for you, because your traces are empty (no device connected). If you have incoming data, the graphs would have switched to power. You can control the behavior in the settings:
[image]

YuryW, 2022-10-25:
It's clear. A lot of the behavior of the application depends on whether the device is connected...
Let's close these questions then...

jankae, 2022-10-25:
I'll increase tests without the device connected in the future. Your case is also special because the only cases you have on the graphs are empty. If they would contain some points (e.g. imported from a touchstone file), the graph behavior would probably be a bit more intuitive.

closed 2022-10-25 by jankae