jankae/LibreVNA #157: GUI_Windows-v1.5.0-2022-10-24-12-10-14 interface crashes

Chorbu, 2022-10-24:
### LibreVNA Version

GUI_Windows-v1.5.0-2022-10-24-12-10-14 latest update interface crashes on startup with popup 

LibreVNA-GUI.exe - System error.
Unable to continue code execution because the system has not detected libusb-1.0.dll . To fix this problem, try reinstalling the program.

Same issue than YuryW

Thank you Jan. 

### Steps to reproduce

LibreVNA startup no gui appear, just popup with DLL missing.

### Expected behavior

correction of this bug and App starting

### Extra information & Setup and Calibration files

no files attached

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

Chorbu, 2022-10-24:
"Thanks to all the creators of the application, device and testers" Yes i'm entirely agree with you ! Device, software are awesome !!  Thank you Jan and all ! Patrick

YuryW, 2022-10-24:
Problem solved. On the forum groups.io/ Jankae posted the required dll (only this one helps!). Other versions of this dll, taken from the Internet, do not help. Copied it to the application folder and all commits now run on W7 at least.
[libusb-1.0.zip](https://github.com/jankae/LibreVNA/files/9855629/libusb-1.0.zip)

jankae, 2022-10-24:
Can you try with d2eb15c as well, please? (download here: https://github.com/jankae/LibreVNA/actions/runs/3315943089). If I fixed everything correctly, that one should start without adding the .dll (but I am currently not on Windows, so I can not check)

YuryW, 2022-10-24:
> Can you try with [d2eb15c](https://github.com/jankae/LibreVNA/commit/d2eb15c776183ba8c86e884a986658796bea520e) as well, please?

For some reason, the download process is not completed. Can not download...(

Chorbu, 2022-10-25:
Hello Jan,

I can't download the latest version as mentionned https://github.com/jankae/LibreVNA/actions/runs/3315943089).

Have a good day. Best regards
Patrick

Chorbu, 2022-10-25:
neither other olds versions on the website.

YuryW, 2022-10-25:
> I can't download the latest version as mentionned

Good morning Patrick. 
Try using the right mouse button Save link as.
In the old fashioned way, I also did not succeed (see above for difficulties).
Best regards.

closed 2022-10-25 by jankae

YuryW, 2022-10-26:
From 10/26/2022, downloading from github by clicking the left mouse button works again.

Chorbu, 2022-10-26:
Yes, it works again. Thank you Jan for solving problem with missing DLL. Sorry, i can't say you "thank you" before because the issue was closed and i'm not familiar with Github (i didn't know how to answer you). I just discover that i can write into closed issues !

Thank you Jan and YuryW
Regards Patrick