jankae/LibreVNA #158: Zero SPAN previous state

Chorbu, 2022-10-30:
### Summary

Hello Jan,

Is it possible to have the possibility to return on previous state (Start, stop, span.. with previous sweep) when zero span is selected ? Maybe on the Same button or other way ?

Thanks in advance

best regards
Patrick

### Paths to implement it

add functionnality

### Extra information

no other information

jankae, 2022-10-30:
The previous sweep frequencies are not saved anywhere. Clicking zero span does not enter a special mode, all it does it set the start and stop frequency to the center frequency (you can also do that manually, it would have the same effect). So no, for now it is not possible to revert back to the previous span.

It probably is possible to create a Ctrl+Z/Ctrl+Y functionality for the span but I am not sure if I am going to do that. If you want to quickly toggle between zerospan and another span, you can create two VNA tabs in the top right corner and set them up with different spans/graphs/markers/everything. Then you can just toggle between these tabs.

Chorbu, 2022-10-30:
Hello Jan,

Ok, effectively, i 'm not used to create another VNA tab, i will try this way. I noticed that you are developing V2 of VNA ?

Thank you and have a good sunday Jan,
Patrick

jankae, 2022-10-31:
> I noticed that you are developing V2 of VNA ?

There is a concept for it and I have a mostly finished schematic and a placed but unrouted layout. The problem is that it is absolutely impossible to build because I can't get all the required parts at the moment. So I wouldn't bet on it being ready any time soon, sorry.

YuryW, 2022-10-31:
> There is a concept for it and I have a mostly finished schematic and a placed but unrouted layout. The problem is that it is absolutely impossible to build because I can't get all the required parts [...]

Good day everyone!
Jankae, what if, in order to increase the decoupling between ports, two new interblock screens are made in the microcircuit placement plane, as in other places (see photo)?
[image]

The difficulty will probably only be to jump to the other side of the copper plating with a jumper in order to bypass these new screens from below in another layer of the printed circuit board in a wide strip.
Best regards.

jankae, 2022-10-31:
I have already made that exact modification, see the layout_6_layers branch. It helped a little bit (and that layout is used in manufacturing of the LibreVNA now) but it wasn't a huge improvement in port isolation.

jankae, 2022-11-20:
Looks like this is solved to me, I'll close the issue.

closed 2022-11-20 by jankae