jankae/LibreVNA #159: Input the load resistance with decimal places

dampf123, 2022-11-01:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.1-5ace021e4
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

- open menue Calibration Kit Coefficients
- open Edit Load Standard

### Expected behavior

I am not able to input decimal places for the load resistance. Only integer.

### Extra information & Setup and Calibration files

.

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

jankae, 2022-11-01:
Thanks, I have increased the displayed number of digits for all calibration standards: aaea3cb

dampf123, 2022-11-01:
Hello Jan,
thanks for the very qick response.

Regards, Klaus

closed 2022-11-01 by dampf123