jankae/LibreVNA #160: Alpha 1.5.0 Crash Windows

Currybrandon, 2022-11-03:
### LibreVNA Version

Windows is suspending and crashing the Alpha build when it sees a LibreVNA connected to PC. Have been unable to keep the program open to even revert the FW change to the LibreVNA, our LibreVNA is now DOA as no other PC can open the APP to revert the FW back. 

### Steps to reproduce

Opening LibreVNA on Windows PC, latest security updates on PC> 

### Expected behavior

Application should be able to open with a libreVNA connected and not suspend itself in task manager. 

### Extra information & Setup and Calibration files

None

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

jankae, 2022-11-03:
That doesn't sound good, but there is hope. I have had reports about crashes from other people as well and it took a while to make improvements because I didn't experience any myself, so debugging was a bit more difficult than usually.
Here are my suggestions, if you have not already tried that:
1. Use the latest version. The 1.5.0-alpha.1 release certainly has bugs that can cause a crash. You can download compiled versions for each commit as well, e.g. [this one](https://github.com/jankae/LibreVNA/actions/runs/3388570145) for the most up to date version (just scroll down on the page and download the Windows version)
2. Start the GUI with the VNA disconnected, switch to the Signal Generator mode, then connect the VNA. If your crash is caused by the same bug as the others, this should be a valid work-around. From there, you can downgrade the firmware to 1.4.0 again and start using the 1.4.0 GUI
3. Use the 1.4.0 GUI to downgrade the firmware. You will get a warning about mismatching protocol versions and you won't be able to see any sweep data, but the firmware update protocol has not changed and should work just the same.

jankae, 2022-11-20:
Any updates from your side? I'd be happy to help with getting back to a running system if you haven't figured it out yourself by now.

jankae, 2022-11-30:
I'll go ahead and close this. Feel free to reopen if you have new information or any updates on the situation