jankae/LibreVNA #161: When the scan range is changed, point parasitic changes in the dynamic range appear.

YuryW, 2022-11-15:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.1-a29c32355
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

### Steps to reproduce

The application is launched with default settings.
The LibreVNA device is connected.
[image]
Change the number of points to 2001.
[image]
Change the minimum scan frequency to 500 MHz.
[image]
Clear parasitic bursts are visible on S21 (5 pieces) and S12 (so far 1 piece).
Now we change the upper scan frequency to 2.5 GHz.
[image]
We already get more than 10 spurious points on S21.
Reduce the IF bandwidth to 200 Hz. Almost the same spurious emissions appear on S12.
[image]

### Expected behavior

Is it possible to get rid of these impulse pickups in the program settings? Or do you need to change something in the software?

### Extra information & Setup and Calibration files

No.

jankae, 2022-11-16:
Thanks for the report, I get very similar spikes (actually also on S12 before changing the IF to 200Hz). Interestingly, there is some difference between the VNA sweep and the manual control dialog. For example:
Zero span sweep at 1.111GHz (the location of a spike):
[image]
Signal level of about -66dB (similar to the spike).
The same settings with manual control:
[image]
Much better S21 result of about -100dB.

This will need some more debugging to determine the cause of the spike (most likely: aliasing of other frequencies into the ADC), I'll get back to you when I have more results.

jankae, 2022-11-16:
Some more info, at least I determined the cause of (most) spikes: It is 1.LO feedthrough mixed with harmonics of the 2.LO in the second stage mixer. Here with the example for the spike at 1.111GHz:
1.IF = 62 MHz -> 1.LO = 1173MHz
2.IF = 250kHz -> 2.LO = 61.75MHz -> 19th harmonic = 1173.25MHz

The second mixer now sees a 1173MHz input and also has some content on its LO at 1173.25MHz -> output 250kHz which the ADC interprets as a valid signal.

Unfortunately, there isn't really anything I can do in the software about that. Here are your possible work-arounds:
1. Shift your 1.LO or 2.LO in the preferences:
[image]
This will not fix the problem but it will move the spikes to other frequencies. Could be a solution if you are only interested in a narrow frequency range and want to get rid of a spike in there.
2. Add a median filter to the traces. This will hide the spikes if they are narrow enough. But of course it would also hide actual narrow signals.
3. Add a lowpass to the 2.LO. The PCB already has the footprints for that but they are not populated:
[image]
[image]

jankae, 2022-11-16:
> And even with an IF width of 10 kHz, once every few seconds in the very upper part of the range, a rather significant spurious impulse (burst) S21 and S12 slips through. (On IF 5 kHz and 20 kHz too, but less often).

Could that peak be at 5.928GHz? If so, please check that the following option is enabled: 
[image]

YuryW, 2022-11-16:
Good afternoon Jankae. Thanks for the thorough answer.

> The second mixer now sees a 1173MHz input and also has some content on its LO at 1173.25MHz -> output 250kHz which the ADC interprets as a valid signal.

It turns out that the excessive broadband of the second mixer is to blame?

> 3\. Add a lowpass to the 2.LO. The PCB already has the footprints for that but they are not populated:

This option suits me the best. That is, LPF (3 pieces) at 62 MHz should practically solve this problem?

> Could that peak be at 5.928GHz? If so, please check that the following option is enabled:

Yes, this option is enabled. I set the marker mark to the frequency of 5.928 GHz: spurious bursts (once in a few tens of seconds or more often) both at this frequency and above it (quite wide sometimes).
[image]
[image]
[image]

jankae, 2022-11-16:
> This option suits me the best. That is, LPF (3 pieces) at 62 MHz should practically solve this problem? Is the location of these filters the same on a 6-layer LibreVNA board?

Yes, if that LPF attenuates the high harmonics of the 2.LO, it should also reduce the spike amplitude significantly (probably all the way to the noise floor for most frequencies). There could still be spikes with other causes but this should take care of most of them. The component location has not changed in the 6-layer board.

> I set the marker mark to the frequency of 5.928 GHz: spurious bursts (once in a few tens of seconds or more often) both at this frequency and above it (quite wide sometimes).

Unfortunately this is something that I can't replicate. I have added an option to show the raw data from the VNA receivers and would like to look at your raw data coming in. Could you please use the latest GUI and perform the following steps?
1. Enable raw data capturing in the preferences:
[image]
After this, you will have eight additional live capture parameters available (RawPortXStageY[Ref]).
2. Create a new trace for each of these parameters (trace name isn't important, but you must select the correct parameter in the traces setup)
[image]
3. Stop the sweep when the peak is present
4. Export all eight raw data traces as .csv (with magnitude and phase) and attach that csv file

YuryW, 2022-11-16:
> 4\. Export all eight raw data traces as .csv (with magnitude and phase) and attach that csv file

[image]
[RawPortStage_1.csv](https://github.com/jankae/LibreVNA/files/10024154/RawPortStage_1.csv)
[image]
[RawPortStage.csv](https://github.com/jankae/LibreVNA/files/10024158/RawPortStage.csv)

jankae, 2022-11-16:
Thank you very much for all the data. There is something in there that can explain the spike, although I still have no satisfying overall explanation: Your reference signal is almost completely missing for two frequency points. Without a reference signal, all receivers essentially only see noise and you get about 0dB for all S-parameters.
Here as an example in one of the raw data traces:
[image]
Don't mind the large absolute dB values, what matters here is the drop of about 90dB at the frequencies of the spike. A bit of background on what happens if you enable "suppress invalid peaks":
- The 1.LO PLL has a limited frequency resolution, especially close to integer multiples of its reference frequency
- Depending on the frequency deviation and the selected IF bandwidth, the 2.IF might fall outside of the IF bandwidth, this would result in something like I am seeing in your raw data
- With "suppress invalid peaks" enabled, I calculate these problematic frequencies in the VNA and adjust the 2.LO to move the 2.IF back into the IF bandwidth

So far, I can think of two possible problems:
1. For some reason, this movement of the 2.LO does not happen for you. I think this is unlikely because it is entirely software based (and not a new feature, so it definitely is in the firmware you are running).
2. Changing the 2.LO requires a PLL reset in the Si5351c to align the phases of all 2.LO signals again. Maybe your copy of the chip takes longer for that to lock? I don't think that parameter is specified in the datasheet and I have based the waiting time on experimental results only. If that PLL fails to lock in time, your 2.LO is either completely missing or at the wrong frequency, which would also explain your measurement results. To test this theory, please try this firmware file: 
[LongerSettlingTimeFirmware.zip](https://github.com/jankae/LibreVNA/files/10025542/LongerSettlingTimeFirmware.zip)
It has an extended the settling time between changing the 2.LO and starting the sampling process.

YuryW, 2022-11-16:
> 2\. Чтобы проверить эту теорию, попробуйте этот файл прошивки: 
>     [LongerSettlingTimeFirmware.zip](https://github.com/jankae/LibreVNA/files/10025542/LongerSettlingTimeFirmware.zip)

Thank you Jankae. This firmware with increased time led to the desired result. In 4 minutes plus 15 minutes, no old outliers appeared at all. But in a strange way, after 4 minutes (device temperature 30 degrees C), the GUI crashed when the traces "froze". After clicking on Stop/Run, the following happened: after Run, there are no traces, but the mark readings changed within the same limits. And after Stop, the traces appeared in the stop mode.
[image]
[image]

After restarting the software, everything worked properly and has been running for 25 minutes without any complaints.

Perhaps my LibreVNA device behaves a little differently than the others, since its operating temperature does not usually exceed 40 degrees C (about 14 ... 15 degrees above ambient temperature). There are two large heatsinks on top and bottom of the device (no fan).

It looks like the 5.928 GHz spike issue has been resolved. At least for me with this firmware.

jankae, 2022-11-16:
Thanks for the confirmation, I'll merge the change into master soon. I will probably get back to you with another firmware to test at some point - I can't really test the change this my Si5351c seems to lock a lot faster than yours.

> the GUI crashed when the traces "froze"

This is likely unrelated but just to confirm: with "crash" you are just referring to the frozen/disappearing traces and the GUI keeps running? I now that there is some issue with the software there, I just can't track it down. But if you happen to find a way to replicate that problem, I'll try to fix it.

YuryW, 2022-11-16:
> But if you happen to find a way to replicate that problem, I'll try to fix it.

Thank you, if the problem of freezing appears, I will definitely let you know.
So far, she has been only once and has not appeared again (for half an hour of work, everything is fine).
Best regards.

jankae, 2022-11-17:
There is now a check for PLL lock before starting the sampling (instead of just waiting a predetermined amount of time). Please check if this version has no 5.928GHz spike with your device as well: https://github.com/jankae/LibreVNA/actions/runs/3488492187

YuryW, 2022-11-17:
> Please check if this version has no 5.928GHz spike with your device as well: https://github.com/jankae/LibreVNA/actions/runs/3488492187

Good evening Jankae. Tested this version of software and firmware:
LibreVNA Version (64 bit): 1.5.0-alpha.1-69b1d95ab
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64
The problem at 5.928GHz seems to be solved. There were no parasitic emissions found. But when testing harder 4001 points and 50 kHz IF, there were spikes at about 1 GHz, 2 GHz and a few more frequencies only on the S21! And for some reason, on S12 at IF 50 kHz, the noise level is much less, well, there are no parasitic bursts.
[image]
[RawPortStage_17_4001ps_20kHz.csv](https://github.com/jankae/LibreVNA/files/10034555/RawPortStage_17_4001ps_20kHz.csv)
[image]
[RawPortStage_17_4001ps_50kHz_1.csv](https://github.com/jankae/LibreVNA/files/10034556/RawPortStage_17_4001ps_50kHz_1.csv)
[image]
[RawPortStage_17_4001ps_50kHz_2.csv](https://github.com/jankae/LibreVNA/files/10034566/RawPortStage_17_4001ps_50kHz_2.csv)
[image]
[RawPortStage_17_4001ps_50kHz_3.csv](https://github.com/jankae/LibreVNA/files/10034571/RawPortStage_17_4001ps_50kHz_3.csv)
[image]
[RawPortStage_17_4001ps_50kHz_4.csv](https://github.com/jankae/LibreVNA/files/10034577/RawPortStage_17_4001ps_50kHz_4.csv)
[image]
[RawPortStage_17_4001ps_50kHz_5.csv](https://github.com/jankae/LibreVNA/files/10034588/RawPortStage_17_4001ps_50kHz_5.csv)
And today, an attempt to check the box "Capture raw receiver values" led to the software freezing again. Something to do with the default calibration mismatch with the default settings. But it was not possible to repeat the situation, so there is no concrete data on this situation yet.

YuryW, 2022-11-18:
Perhaps this information will help:
[image]
[RawPortStage_19_4001ps_5_6GHz_30kHz.csv](https://github.com/jankae/LibreVNA/files/10045522/RawPortStage_19_4001ps_5_6GHz_30kHz.csv)
[image]
[RawPortStage_19_4001ps_5_6GHz_40kHz.csv](https://github.com/jankae/LibreVNA/files/10045523/RawPortStage_19_4001ps_5_6GHz_40kHz.csv)
[image]
[RawPortStage_19_4001ps_5_6GHz_47kHz.csv](https://github.com/jankae/LibreVNA/files/10045525/RawPortStage_19_4001ps_5_6GHz_47kHz.csv)
[image]
[RawPortStage_19_4001ps_5_6GHz_48kHz.csv](https://github.com/jankae/LibreVNA/files/10045526/RawPortStage_19_4001ps_5_6GHz_48kHz.csv)
[image]
[RawPortStage_19_4001ps_5_6GHz_50kHz.csv](https://github.com/jankae/LibreVNA/files/10045530/RawPortStage_19_4001ps_5_6GHz_50kHz.csv)
Best regards.

jankae, 2022-11-19:
Looks like another PLL issue to me. I could reproduce it but not with every copy of the LibreVNA I have. This time it is the MAX2871 PLL (for the source and the 1.LO). I already monitor the lock output and wait with sampling until the PLL claims to have locked (I actually wait for 20us after the lock has happened to allow for everything to settle).

Turns out that the MAX2871 sometimes keeps the lock output asserted when making a small frequency change. Maybe because the charge pump only needs to make a small adjustment? Not sure. But whatever the cause, I couldn't change the behavior with any of the registers that influence lock detect behavior.
The only thing I could come up with is increasing the settling time from 20us to 60us. This solved the issue for me but of course at the cost of reduced sweep speed. Please check if this helps in your case as well with [this](https://github.com/jankae/LibreVNA/actions/runs/3504081971) firmware.

By the way: you only get the spikes in S21 and S12 because they are sampled first. By the time S12 and S22 are sampled, the PLLs have properly locked.

YuryW, 2022-11-19:
> Please check if this helps in your case as well with [this](https://github.com/jankae/LibreVNA/actions/runs/3504081971) firmware.

Good evening Jankae. Yes, everything worked out. A new firmware is required. Only one updated software did not help, but with the new firmware everything is OK.
[image]
[image]
[image]
I have the impression that sampling (its order) also affects the level of spurious emissions due to 2.LO harmonics.
[image]
In any case, until I put filters on 2.LO, you can deal with parasitic bursts by changing the number of points, as shown below:
[image]
[image]
Best regards.

YuryW, 2022-11-19:
Tried other edge cases. And here is the result.
[image]
[RawPortStage_20_4499ps_1MHz_6GHz_50Hz.csv](https://github.com/jankae/LibreVNA/files/10048093/RawPortStage_20_4499ps_1MHz_6GHz_50Hz.csv)
[image]
[RawPortStage_20_4499ps_1MHz_6GHz_10Hz.csv](https://github.com/jankae/LibreVNA/files/10048095/RawPortStage_20_4499ps_1MHz_6GHz_10Hz.csv)
As if at some point, the noise was inverted (instead of the bottom, it began to count up).

jankae, 2022-11-19:
> Tried other edge cases. And here is the result.

Yes, you have run into the limit of the "suppress invalid peaks" there. The frequencies at which the 2.LO needs to be shifted is calculated in advance and stored in a lookup table with 500 entries. With such low IF bandwidths and so many points, the 2.LO needs to be shifted more often than that and afterwards the peaks can no longer be suppressed. You should see corresponding warnings in the device log ("PLL deviation of 743Hz for measurement at 5997332592Hz, will cause a peak").
The easiest solution of just increasing the size of the lookup table is not possible due to the limited amount of RAM. I'll have to come up with something a bit more clever than that.

YuryW, 2022-11-19:
Yes, I found warnings. Example: "PLL deviation of 41Hz for measurement at 4142976222Hz, will cause a peak". Somehow I didn't look at the logs before.
As an option to programmatically, probably, limit the width of the IF for a certain number of points? For such a number of points, frequency range and 100 Hz IF, everything works normally.
[image]

jankae, 2022-11-20:
I think I found a cleaner solution: get rid of the lookup table altogether and calculate the required shifts of the 2.LO right when they are needed. When I initially implemented the lookup table I thought it would save time but when shifting the 2.LO I have to wait for it to lock anyway. Any additional calculation before that isn't really that critical.

Another firmware to test: https://github.com/jankae/LibreVNA/actions/runs/3505746388 (no changes to the GUI, updating the firmware is enough)

By the way: thanks a lot for all the extensive tests/screenshots/data :)

YuryW, 2022-11-20:
> Another firmware to test: https://github.com/jankae/LibreVNA/actions/runs/3505746388 (no changes to the GUI, updating the firmware is enough)

It just worked out great. Thank you very much. There are small incomprehensibility for me (jumps in phase and amplitude). Circled them. Perhaps this is not important, but just in case, and suddenly it will help, if anything. I think that we can now safely close the issue.
[image]
[RawPortStage_20_4499ps_1MHz_6GHz_10Hz_new.csv](https://github.com/jankae/LibreVNA/files/10048980/RawPortStage_20_4499ps_1MHz_6GHz_10Hz_new.csv)
[image]
[RawPortStage_20_4499ps_5GHz_6GHz_10Hz_new.csv](https://github.com/jankae/LibreVNA/files/10048981/RawPortStage_20_4499ps_5GHz_6GHz_10Hz_new.csv)
[image]
[image]
Best regards.

jankae, 2022-11-20:
> But with this firmware, what are the restrictions on the number of points now?

Still the 4501 point limit (limited by the amount of RAM in the FPGA). But the GUI already includes a segmented sweep option which automatically splits the sweep into multiple smaller sweeps (on the hardware level). This is completely transparent for the user, the only thing you might notice is a short pause in the sweep when the next segment is set up in the hardware. With this feature enabled, the point limit is 65535 (and could also be increased but I don't think anyone needs more than that).

> I think that we can now safely close the issue.

Okay, I'll go ahead and close this.

closed 2022-11-20 by jankae

YuryW, 2023-01-05:
> Some more info, at least I determined the cause of (most) spikes: It is 1.LO feedthrough mixed with harmonics of the 2.LO in the second stage mixer. Here with the example for the spike at 1.111GHz: [...]

Hi Jankae,
Designed the LP filter for 2.LO based on the PCB layout of 6 layers of LibreVNA. Used QucsStudio. In my opinion, I used QucsStudio to the maximum in order to take into account all the parasitic elements that distinguish the filter from the ideal one in this frequency range (it is in the upper right corner of the diagram). The 5.1 pF capacitor improves the dynamic range at 6 GHz by about 10 dB (until I started using it). Capacitors must be in parallel of two of a smaller value, and not one of a larger one, so that filtering (dynamic range) is better according to QucsStudio.

[image]

Who wants to repeat this filter - pay attention to the marking of the applied capacitors and inductances from Murata. With other capacitors and inductors, the frequency response will be different.
All harmonics above 700 MHz are filtered better than 100 dB.
Here are the manifestations of harmonics of the local oscillator without a 62 MHz filter (see above (earlier) in the text of this problem.

[image]

And here are new measurements with similar old settings already with LPF 62 MHz applied. At S21, part of the parasitic bursts noticeably decreased, especially above 1.2 GHz, but at IF 1 kHz at S12, their number seems to have increased. In theory, these bursts should no longer be above 700 MHz.

[image]

[image]

YuryW, 2023-01-05:
> The second mixer now sees a 1173MHz input

If Jankae doesn't mind, then here or (probably better there) on the forum, I will later consider the pros and cons of other filters that come after the first and second mixer using QucsStudio.
Best regards.

YuryW, 2023-01-08:
I will offer an easy to remember way to solve the problem, when in such a case - the initial data: VNA, Start frequency 500 MHz, end frequency 2.5 GHz, number of points 2001, IF bandwidth 100 Hz, spikes appear, as in the picture below:

[image]

You need to shift the initial scan frequency of the VNA up or down by 250 kHz (IF2). In the vast majority of cases, such a change is acceptable for measurements and is practically not noticeable. But there is an effect:

[image]
[image]

Whether this always works will be shown by the experience of dear readers of the topics discussed here on LibreVNA (a wonderful assistant for amateurs and radio professionals).
Post your results here.

YuryW, 2023-01-10:
VNA mode: start frequency 1 MHz, stop frequency 6 MHz - there are bursts.

[image]

VNA mode: start frequency 1.25 MHz (750kHz), stop frequency 6 MHz - no bursts.

[image]
[image]