jankae/LibreVNA #162: Parameters for PCB manufacturing + gerbers

WojtaCZ, 2022-11-19:
### Summary

Could you please provide some parameters that the PCB should be manufactured with plus gerbers generated from the eagle project?

### Paths to implement it

Usefull parameters:
- Stackup, either table with necessary values or manufacturer and their standard
- Special requirements (blind vias, POVF vias etc.)
- Basic parameters (pcb size, number of layers...)

Exported gerbers ready for manufacturing for those who don't work with eagle.

### Extra information

 I guess the parameters can be found out from the eagle project but it still would be nice to have them listed somewhere for non Eagle users.

jankae, 2022-11-19:
Sure, I have added the required files here: https://github.com/jankae/LibreVNA/tree/master/Hardware/Gerber

> Stackup, either table with necessary values or manufacturer and their standard

Screenshot from Eagle DRC is included.

> Special requirements (blind vias, POVF vias etc.)

Nothing special, just normal through hole vias. I am not sure what you mean by POVF vias.

WojtaCZ, 2022-11-19:
Perfect, thank you very much :)

closed 2022-11-19 by WojtaCZ