jankae/LibreVNA #167: Add to the Matching Network menu the parallel defined in the touchstone file (s2p).

YuryW, 2022-12-02:
### Summary

Many manufacturers of L and C publish their s2p parameters (touchstone file). In this edition of the LibreVNA application (LibreVNA Version (64 bit): 1.5.0-alpha.2-35cdfa592), it is not possible to insert a parallel element into the matching circuit, specified by the touchstone file. Is it possible to change this? Add to the Matching Network menu the parallel defined in the touchstone file (s2p).
[image]

### Paths to implement it

No

### Extra information

No

jankae, 2022-12-02:
Should be possible but I can't find the required math right now. If you can point me to that, it should be straight forward to implement.

YuryW, 2022-12-02:
Good evening Jankae.
I'll try to formulate my logic. If I'm not mistaken, then perhaps it will help. We need to obtain conductivity in a similar way to the already familiar (judging by the result already implemented in LibreVNA) connection in parallel with the capacitance C. Among the conductivity parameters Ynn, we are interested in Y11 or Y22, since they are obtained by a short circuit on the opposite side of the four-pole, which is present when our four-pole is connected in parallel in the matching circuit.
Formulas for recalculating Y-parameters from S-parameters of a quadripole can be taken from here: 
https://www.spectrum-soft.com/news/winter2001/conv_spar.shtm 
D = ((1+S11)*(1+S22)-S12*S21)
y11 = ((1-S11)*(1+S22)+S12*S21) / D
y22 = ((1+S11)*(1-S22)+S12*S21) / D. 
As a result, we will get a complex conductivity similar to the complex conductivity of capacitance C. And its integration into the DUT matching chain has already been worked out in LibreVNA. If complex resistance Z is required, then it is the opposite of complex conductivity Y. In any case, the correctness of the implemented algorithm can be checked in a simulator, for example, QucsStudio (https://dd6um.darc.de/QucsStudio/qucsstudio.html).
And further. So far, I'm unable to negotiate the DUT chain of the downloaded touchstone file when the LibreVNA device is not connected. Here are some screenshots. 
[image]
[image]
[image]

Maybe I didn't read the manual carefully. I would like to have a clear algorithm of actions, and so that when changing the matching circuit, and after pressing OK in the Matching Network menu, the result would be immediately visible on the graphs.

jankae, 2022-12-02:
> I'll try to formulate my logic

Thanks, your math looks reasonable to me. I'll probably try myself at the implementation soon.

> So far, I'm unable to negotiate the DUT chain of the downloaded touchstone file when the LibreVNA device is not connected.

There is a bug in the de-embedding for the matching network, it tries to de-embed as many ports as the connected VNA has (or, in case of no connected VNA, 2 ports). You only have the de-embedding configured for one port, so you can only supply one trace. The crash happens, when the algorithm starts looking for the other traces (which it shouldn't need here). I am working on a solution.

> I would like to have a clear algorithm of actions, and so that when changing the matching circuit, and after pressing OK in the Matching Network menu, the result would be immediately visible on the graphs.

There are two different scenarios:
1. You are de-embedding incoming VNA data. This should already work as you described here (new data on graphs will use the changed matching network after you change it).
2. De-embedding already existing traces. This will not work in this way and I don't think that is likely to change. De-embedding already existing traces modifies the trace data itself. This is a non-reversible operation. Afterwards, the GUI has no knowledge of which traces have been de-embedded, so it can't adjust them if you change your de-embedding setup.

YuryW, 2022-12-03:
Good day.
For a simple user of the LibreVNA application, I will explain how the DUT negotiation of the previously saved touchstone file works now (with no LibreVNA device connected). LibreVNA Version (64 bit): 1.5.0-alpha.2-35cdfa592.
First, in the open Matching Network menu, you need to insert all the elements you need into the DUT matching circuit and enter their parameters. 
[image]

Only after that import the touchstone file. Then a menu will appear with a suggestion for de-embedding imported tracks. Check the box for deembedding. 
[image]

After that, on the graphs, we add the corresponding traces with checkmarks and they appear already taking into account the matching circuit. 
[image]

With any changes in the matching circuit, all procedures must be repeated. Also, for this matching network, you can import other traces with their subsequent addition to the graphs.
[image]

Best regards.

YuryW, 2022-12-03:
Compared connecting a 1 nH inductor in series on port 1 in QucsStudio and LibreVNA. So far, the same result is not obtained after matching, although everything worked the same way before turning on the inductance.
[image]
[image]
[image]
[image]
[image]

jankae, 2022-12-03:
I also noticed a couple of other bugs in the matching network de-embedding. Mostly introduced by the recent changes for more flexible port arrangements. I'll fix them and do some more tests but it will take a couple of days, I am quite busy at the moment.

YuryW, 2022-12-06:
Good afternoon.
In LibreVNA Version (64 bit): 1.4.0-7f6b78992, not only matching network worked great, but it also had the ability to work on two ports at once. Is it possible to return all this again in the new edition? It will be easier and cooler, in my opinion. ))
[image]
[image]

jankae, 2022-12-06:
>  it also had the ability to work on two ports at once or with any of the ports separately, placing the elements just in the right places. Is it possible to return all this again in the new edition?

In 1.5.0 you can add multiple matching networks. If you add one for port 1 and one for port 2, this should give you the same options as in 1.4.0 (but of course you can't view both matching networks at the same time). This is more flexible because it allows for matching networks on any number of ports (1.4.0 was limited to two ports).

It also makes the calculations a bit more complicated (I am using ABCD-parameters internally for the matching network and haven't fully figured out yet how to concatenate networks with more than two ports). But once the bugs have been fixed, this should be working correctly just as in 1.4.0.

YuryW, 2022-12-06:
> (but of course you can't view both matching networks at the same time)

Good evening jankae.
Viewing and configuring just two matching circuits is the privilege of VNA, which have the ability to measure S11, S21, S12, S22 at once, as it was in the first version of the LibreVNA device. All competitors had to reconnect the DUT (change the input with the output). If increasing the number of inputs by combining several LibreVNA devices into one measuring complex leads to the loss of the former advantage in viewing two or more matching chains at different inputs at once, then this, in my opinion, is a deliberate rejection of advantages over competitors.
Adjusting one input changes the settings of another and sometimes it is important to be able to see and adjust the other input as well. And how is this option - to leave the option of matching circuits with two ports, as it was in LibreVNA 1.4 and be able to select multiport settings for several devices, but then with viewing and configuring each port separately?
Is it possible to leave the old option when adding a new one to the menu?
That is, within the same LibreVNA device, it remained possible to configure and see two ports at once.
Otherwise, it turns out that several devices are combined into a single measuring complex with the ability of each to measure only S11 and S21.

YuryW, 2022-12-06:
> I am using ABCD-parameters internally for the matching network

Are ABCD parameters used to calculate the cascading of matching circuit elements? It is common for S-parameters to use a transformation to T-parameters, matrix multiplication when cascading, then the inverse transformation.
(https://www.rfcafe.com/references/electrical/s-h-y-z.htm)

jankae, 2022-12-07:
> Are ABCD parameters used to calculate the cascading of matching circuit elements? It is common for S-parameters to use a transformation to T-parameters, matrix multiplication when cascading, then the inverse transformation.

Both ABCD and T parameters can be used for that: https://electronics.stackexchange.com/questions/620767/are-t-parameters-the-same-thing-as-abcd-parameters
I chose ABCD parameters because they provide a simple way to implement shunt and series impedances: https://en.wikipedia.org/wiki/Two-port_network#Table_of_transmission_parameters

> Viewing and configuring just two matching circuits is the privilege of VNA, which have the ability to measure S11, S21, S12, S22 at once, as it was in the first version of the LibreVNA device.

And this will continue to work in all future versions. I think there might have been a misunderstanding: you can still configure a matching network on both ports in 1.5.0, you just have to add two (one for each port) and don't have the convenience of editing (what I meant by "viewing") both networks at the same time.

Two matching networks configured in the de-embedding dialog:
[image]
First network is for port 1:
[image]
Second network is for port 2:
[image]

This would result in the same transformations as with the two-port matching network in 1.4.0. Of course it doesn't at the moment because of the bugs in the math...

Disadvantage: you'll have to configure the two networks one after the other
Advantage: the matching network will work for any number of ports

YuryW, 2022-12-07:
> I chose ABCD parameters because they provide a simple way to implement shunt and series impedances: https://en.wikipedia.org/wiki/Two-port_network#Table_of_transmission_parameters

This is clear.

> This would result in the same transformations as with the two-port matching network in 1.4.0.

That is, with such a setting of the input matching circuits for inputs 1 and 2, I can observe on the graphs all traces S11, S21, S12, S22 at the same time, taking into account both matching circuits in them. It’s just that with such a given scheme for matching individual ports, it’s somehow not obvious that in this case S21 and S12 will be calculated ...))

jankae, 2022-12-07:
> That is, with such a setting of the input matching circuits for inputs 1 and 2, I can observe on the graphs all traces S11, S21, S12, S22 at the same time, taking into account both matching circuits in them.

Yes, all affected S parameters are updated, even if the matching network is only defined at one port.

> And if the second port is port 3, then to calculate S31, S13 (without a phase, if sNp is not from the touchstone file), it is necessary to create a custom port 3 circuit (even without matching elements in the circuit)? Or will the calculation between ports always be automatic?

There is no need to add a matching network if it will be empty. In this case, S31 and S13 (and also S33) would be updated automatically if a matching network is configured at port 1.
All S parameters are assumed to have a phase. Even if they were created without phase information (e.g. by importing a csv file without a phase column), the phase is assumed to be zero and might be altered by the de-embedding.

By the way: I believe I found the bug in the matching network de-embedding. It was an uninitialized reference impedance when passing the trace data to the de-embedding module. As a consequence all the conversions between S and ABCD parameters resulted in wrong values.
With 77f73d3e I am able to reproduce your earlier example and am getting the same graphs as Qucs:
[image]

YuryW, 2022-12-08:
> By the way: I believe I found the bug in the matching network de-embedding.

Thank you very much, Jankae, for a detailed and informative conversation and for your work on improving LibreVNA. Everything works great. 
[image]

It seems that now the recalculated trace is saved instead of the previous one, and if you press OK again in the new Manual De-embedding menu, 
[image]

the previous de-embedding chain will be recalculated.

[image]

[image]

And in the new edition, the order of actions is no longer important - importing a touchstone file or creating a matching chain can be done in any sequence.

jankae, 2022-12-08:
> the previous de-embedding chain will be recalculated.

This was always the case. The de-embedding changes the trace data, so a second de-embedding will just de-embed the already de-embedded data again. I realize that this may be annoying when de-embedding imported data a lot with changing de-embedding settings (would need to import the same data again to start with the original value). But I don't see an easy way to implement it otherwise.

> And in the new edition, the order of actions is no longer important - importing a touchstone file or creating a matching chain can be done in any sequence.

Exactly, you can either de-embed a touchstone file while importing it or do so later - the result will be the same.

I fixed another bug (wrong order of component calculation), it was barely visible in your test due to the choice of components. If I compare my results to Qucs now, I can not detect any visible differences.

Also, there is now a "defined shunt" option (basically a connection to ground defined by a touchstone file). Your suggested approach to the calculation seems to work fine.
[image]

YuryW, 2022-12-08:
Good afternoon.
Didn't create a new problem. I think it has something to do with this one.
In the current version, the s3p touchstone file is not imported correctly.
Seems like a confusion between MHz and GHz...
[image]

[image]

jankae, 2022-12-08:
Can you please attach the touchstone file?

YuryW, 2022-12-08:
[TCM1-63AX+_unit1.zip](https://github.com/jankae/LibreVNA/files/10185609/TCM1-63AX%2B_unit1.zip)

jankae, 2022-12-08:
That file doesn't have a space between the '#' and frequency unit indication on the option line. I have only seen files with a space there (although the file format specification doesn't explicitly state that it is required) and this breaks my parsing which defaults to GHz. It is not related to this issue but the fix should be simple and quick.

jankae, 2022-12-08:
Should be fixed now

YuryW, 2022-12-08:
> there is now a "defined shunt" option

For importing touchstone s2p and a capacitor, s2p also seems to work.
[image]

[image]

For touchstone s3p and capacitor imports, s2p also works only in order - first the matching chain, then the import. And you can’t add Port 3 in the Manual De-embedding and Matching Network menus. This is probably why the route calculation is slightly different.

[image]

[image]

[sXp.zip](https://github.com/jankae/LibreVNA/files/10186950/sXp.zip)

YuryW, 2022-12-08:
Нет, расчет очень точен. Это я перепутал в QucsStudio порты 2 и 3, а они не совсем идентичны. So for now, the problem is only when importing touchstone s3p - there is no possibility of negotiating on port 3. 

[image]

[image]

Кстати, QucsStudio - это не Qucs, так что "не обижайте" его создателя... ))
( http://qucsstudio.de/)

jankae, 2022-12-08:
> there is no possibility of matching on port 3.

The de-embedding options were limited to the number of ports on the connected device (or 2, if no device is connected). But your experiments show that it might be useful to allow higher port numbers as well when importing other data. The latest commit always allows the user to go up to port 8 for the de-embedding, regardless of whether a device is connected or not.

jankae, 2022-12-08:
> By the way, QucsStudio is not Qucs, so don't "offend" its creator ... ))

Ah, I wasn't aware that these are actually different projects. I found the user interface of Qucs a bit unintuitive when I last tried it, I might give QucsStudio a go

YuryW, 2022-12-08:
> I might give QucsStudio a go

"... QucsStudio is mainly a circuit simulator that has evolved out of the project [Qucs](http://qucs.sourceforge.net/), but isn't compatible with it. The simulation engine is even a complete new creation...."

Qucs is no more. There is a "low frequency" version, now called Qucs S (Spice) and a high frequency version that is very fancy, cool for free versions of QucsStudio.

YuryW, 2022-12-08:
> The latest commit always allows the user to go up to port 8 for the de-embedding, regardless of whether a device is connected or not.

If you do it in sequence - first matching networks in ports, then importing s3p and ticking de-embedding, then the calculation does not take into account the elements in port 3.
[image]

If further in the Manual De-embedding menu you select specific traces:
[image]

then we get this.
[image]

If you first import s3p, and then create matching networks for ports, then the calculation after OK in the De-embedding menu does not occur. And after selecting the traces in the Manual De-embedding menu, the calculation takes place without taking into account the elements in support 3.
And here is the result in QucsStudio:
[image]

[sXp.zip](https://github.com/jankae/LibreVNA/files/10190425/sXp.zip)

jankae, 2022-12-08:
> the calculation does not take into account the elements in port 3

Unable to reproduce. No matter in which order I do the de-embedding (set up the chain first then import or import first and de-embed later), I very clearly get results on all three ports:
Without any de-embedding:
[image]
After de-embedding:
[image]
The matching network is identical for all three ports:
[image]

YuryW, 2022-12-09:
And I have the same result.
But the transfer characteristics do not match with QucsStudio, and the reflection parameters seem to be close. Tweaked your test a bit to get away from very large values ​​in trace results.
[image]

[image]

Need to test more.

YuryW, 2022-12-09:
Yes, and I also noticed that if you don’t like the color on the de-embedding track and change it, then the track, after updating its color, is drawn, but in the form that it had before de-embedding.

jankae, 2022-12-09:
> But the transfer characteristics do not match with QucsStudio, and the reflection parameters seem to be close.

I think I figured it out now. Consider this matching network:
[image]
My internal S or ABCD parameters only support two port networks, so I perform two de-embedding calculations: One uses P1 and P3 of the DUT, the other P1 and P2 (P1 is calculated twice but the result is the same - no issue here it will just get overwritten with the same value again). But this doesn't change S32 and S23. I initially thought that was fine but there is actually another signal path now.
Example for a signal from P3 to P2:
This is transferred through S23, but also through
```
S13*Sxx*S21
```
where Sxx is the reflection from the DUT P1 at the matching network.
Sxx can be calculated as 
```
Sxx = SM22/(1-SM22*S11)
```
where SM22 is the return loss of the matching network at the DUT.

So S23 actually also needs to be updated when de-embedding:
```
S23 = S23 + S13*SM22/(1-SM22*S11)*S21
```

With these changes added to the code I now get identical results as in QucsStudio (I imported the de-embedded S parameters from QucsStudio, they match my calculated traces exactly).

YuryW, 2022-12-09:
In my test, too, everything coincides perfectly with QucsStudio.
(Well, QucsStudio has already come in handy).

[image]
[image]

But in such a test (LibreVNA Version (64 bit): 1.5.0-alpha.2-b67275831) : s2p import and parallel inductance 1 nH on both ports, everything is the same as before.

[image]

[image]

dna35, 2022-12-10:
Impressed!

jankae, 2022-12-10:
Okay, looks like this finally works as intended. Thank you for all the help with the debugging :)
If you find no further problems, I propose to close this issue now (the resetting trace when changing the color is unrelated. I am already on it, but you can also open a new issue for that if you want)

closed 2022-12-10 by YuryW