jankae/LibreVNA #170: Add a non-uniformity measurement of the DUT transfer characteristic to marker calculations.

YuryW, 2022-12-11:
### Summary

Measuring the unevenness of the DUT transfer characteristic (one of the parameters for the compliance of many amplifying or passive devices with the requirements of various levels of standards).
Non-uniformity function parameters:
non-uniformity - there are maxima of deviations in "plus" and in
"minus" from a straight line connecting
markers 1 and 2. Unevenness is determined
as their sum.
[image]

### Paths to implement it

We take mathematics from measuring the distance from a point to a straight line given by two points:
[image]

More math details here:
https://en.wikipedia.org/wiki/Distance_from_a_point_to_a_line

Where x1 is the lower frequency limiter, x2 is the upper frequency limiter, y1 is the marker value at the lower frequency limiter, y2 is the marker value at the upper frequency value, x0, y0 are an arbitrary frequency value and the corresponding marker values in the section between the limiters along frequency. That is, by entering both frequency limiters into the menu, we get five "constants"
y1,
x1,
А = sqr( (x2 – x1) * (x2 – x1) + (y2 – y1) * (y2 – y1) ),
B = x2 – x1,
С = y2 – y1,
and two variables y0 and x0.
And then, when drawing the trace graph, it remains for us to calculate all the deviations of the trace from the straight line using the formula
distance = ( B * (y1 - y0) - C * (x1 - x0) ) / A.
When scanning by frequency, comparing two adjacent positive (and separately negative) values with each other and leaving a larger (and separately smaller, taking into account the sign) value, we obtain the maximum deviation from the straight line in plus (maximum delta +) or minus (maximum delta - ) :
two important parameters and the sum of positive and negative deviation from the straight line as the full swing of the deviation (the third important parameter).
We display the values of these three parameters in the menu (**or only the values of delta + and delta -**).

This is for the case where all trace points between frequency limiters are stored and available for calculation.

### Extra information

No.

jankae, 2022-12-14:
4568be8f adds a non-unity marker type. I didn't use the math you linked because the non-uniformity seems to only take into account the *vertical* deviation from a straight line.

jankae, 2022-12-14:
By the way: could you please send me the link from where you got the main picture in the first post? I think I have seen it before but can't seem to find it anymore

YuryW, 2022-12-14:
So we are engaged in marker (graphic) calculations. That is, only what we have in the form of traces on the graphs, that is, dB and frequency. Therefore, we substitute the data from the traces (dB and frequency) into the formulas and get dB as a result.
It's just that when working with formulas, we abstract that we have dB and frequencies, we work only with numbers, functions, graphs, straight segments ... And when we get the result, it will be in dB, because we worked with numbers that actually were dB.
Or even so, if the hypotenuse of a right triangle is measured in dB, why can't the legs of this triangle be measured in dB and calculated using the Pythagorean theorem?

YuryW, 2022-12-14:
> By the way: could you please send me the link from where you got the main picture in the first post? I think I have seen it before but can't seem to find it anymore

I just raised this question here. Now I'll try to find the original source in English.

jankae, 2022-12-14:
> Therefore, we substitute the data from the traces (dB and frequency) into the formulas and get dB as a result.

No that doesn't work. You need to able to carry units during calculations. In the formula for the perpendicular distance, there would be a problem with the units:
Look at the square root in the denominator: sqrt((x2 - x1)^2+(y2-y1)^2)
x2 and x1 would be in Hz, y2 and y1 in dB. So in units we get sqrt([Hz]-[Hz])^2+([dB]-[dB])^2) or sqrt([Hz]^2+[dB]^2)
Adding two values with different units just makes no sense.

If you look at it in another way, the vertical distance even depends on the ratio of the graph. The formula would calculate the vertical distance only if the graph used the same distance in X for 1Hz as it does in Y for 1dB.

YuryW, 2022-12-14:
In functions, the x-axis is always the dimension of one quantity, such as time, and the y-axis is the value of another quantity, such as voltage.
When calculating in geometry, we only have the coordinates (x,y) , only the numbers by which we build graphs...
On graphs in LibreVNA, we only have x and y coordinates of points. There are no dB and frequencies, only numbers ...

jankae, 2022-12-14:
> There are no dB and frequencies, only numbers ...

Sure, if you want to have the perpendicular distance in pixels that would be the way to go. But that number isn't especially useful.
But you can't just drop the units, do a calculation and then re-apply an arbitrary unit. How do you even decide that dB would be the correct unit to apply afterwards? You are inputting Hz and dB (and the formula is symmetric, you would get the same result when swapping x and y), so could you also say the result is in Hz?

> We can also take the derivative and get the slew rate of the transfer characteristic versus frequency.

Yes, that is just a derivative with respect to frequency and that would have the unit [dB/Hz] (which you could also derive by carrying the unit through the required equations). But I don't think it has anything to do with non-uniformity.

YuryW, 2022-12-14:
Or so. I'm making a screenshot of a transfer characteristic plot in the LibreVNA application. Then I use the program for digitizing graphs, I get other coordinates of all points (of course, the proportions will be saved) and, using the above mathematics, I calculate the distance from the point to the straight line at each point of the transfer characteristic graph. But in fact, I can do this with the x (frequency) and y (dB) coordinates that are already in LibreVNA without any conversions. Is not it so?

> Sure, if you want to have the perpendicular distance in pixels

Why in pixels? We will insert the values ​​of the coordinates (x, y) and all calculations will be in proportion to the distances in x and y. Indeed, on the graphs, we simply take the coordinates of functions given by points (transfer characteristic) and a straight line (a line connecting the coordinates of our limits (x1, y1) - (x2, y2)).

jankae, 2022-12-14:
> We will insert the values ​​of the coordinates (x, y)

You can not do that. It is a fundamental mathematical principle that you can not add values with different units. And that would happen in the formula for a distance between a line and a point if the x and y coordinates do have different units.

Just because you add numbers, the result isn't necessarily meaningful. See also
https://en.wikipedia.org/wiki/Unit_of_measurement#Calculations_with_units_of_measurement
especially this line: "Treat units algebraically. Only add like terms."
and as a more general overview
https://en.wikipedia.org/wiki/Dimensional_analysis#Mathematics
with this statement: "Only commensurable quantities (physical quantities having the same dimension) may be compared, equated, added, or subtracted."

I appreciate that you looked up all the math for point to line distance. In fact, I already have a function for that in the GUI because I need it in other parts of the software:
https://github.com/jankae/LibreVNA/blob/4568be8ff4453650f4c318cc3d113f82f0514095/Software/PC_Application/LibreVNA-GUI/Util/util.cpp#L36

But I am positive that this formula can not be applied here. It doesn't really make sense to define a perpendicular distance here anyway because that distance would change if you display the data differently. For example, if you compress the graph horizontally, the most distant perpendicular point would change. Surely, you don't want to have a calculated non-uniformity that depends on how the data is rendered.

I think it would help if you find the original source of that image for the non-uniformity definition. I am not having much luck with finding much information on that topic.

YuryW, 2022-12-14:
> I already have a function for that in the GUI

Well, yes, tolerance control, when there are Constant Lines on the chart - this is almost the necessary mathematics. Only the coordinates of the ends of such a line are placed by markers on the transfer characteristic. Agree that in both cases the points are given by frequency and dB. And then you need to use any mathematics to calculate the shortest distance between this line and the points on the transfer characteristic.
Moreover, this calculation can be vertical or perpendicular to a straight line. This is an important subtlety. If it is vertical, then the measurements will actually be slightly overestimated, since the hypotenuse is always longer than the leg in a right-angled triangle, and the more the line between the markers is inclined from the horizontal, the greater this error will be. The drawings draw a vertical segment to the straight line, and the description refers to the distance from the measurement point to the straight line (this is a perpendicular). And this can sometimes lead to confusion.
Here is the manual for VNA. See transfer flatness measurement on pages 150-151.
https://rtelecom.net/userfiles/product_files_shared/Planar/TR1300/TRVNA_Operating_Manual.pdf

We can see on the graph the distance between the Constant Lines and the transfer characteristic trace. Why can't we calculate this distance?

jankae, 2022-12-14:
> Agree that in both cases the points are given by frequency and dB

Yes, the points are ultimately defined given in Hz and dB.

> And then you need to use any mathematics to calculate the shortest distance between this line and the points on the transfer characteristic.

Also yes, but with a few important differences (these apply to all calls to Util::distanceToLine):
- I convert the points to a pixel position on the graph. That way, the x and y coordinates have the same unit (pixels)
- I am not interested in the absolute value of the distance. I just want to find the point with the smallest or highest distance. The function gives me a valid distance (since both x and y coordinates have the same unit) but the distance also depends on the size of the graph on screen.

> Moreover, this calculation can be vertical or perpendicular to a straight line.

With the risk of repeating myself: no, you can not calculate a distance in any arbitrary direction on a graph with different units for x and y axes. You can only calculate vertical and horizontal distances. Maybe this is a language barrier, but I really don't know how to explain it much better.
It may seem counterintuitive because you look at a graph, see a line and think "hey, we must be able to determine the length of that line". And we can, *if we want to have the length in pixels* (or cm if we also know the pixel pitch). But that value doesn't help us because it is very much dependent on how the graph is drawn (proportions, overall size).

You can not apply the same formula with different input units for x and y. It would be the same as adding your height and age. What would be the unit/dimension of the result? Sure, you get a number, but the number is completely useless.

> the description refers to the distance from the measurement point to the straight line (this is a perpendicular).

Your linked document actually doesn't mention "distance" at all in reference to flatness. It mentions positive and negative peaks measured from the line between the markers. And in the drawing, the line that indicates these peaks is very clearly vertical and not perpendicular.

But just for fun, here is an example with your proposed formula:
[image]
The line (red) is given by two markers:
1: 407MHz / -4.265dB
2: 395.6MHz / -38.41dB
The furthest point from the line is given by another marker:
3: 400.2MHz / -2.166 dB

The green line indicates the perpendicular "distance". Let's plug the values into the formula:
The line:
```
x1 = 407000000 Hz
x2 = 395600000 Hz
y1 = -4.265 dB
y2 = -38.41 dB
```
The point:
```
x0 = 400200000 Hz
y0 = -2.166 dB
```
... dropping all units and just pretending the result is in dB gives...
```
22.466 dB
```
Which isn't completely off but I hope you can clearly see that the green line is definitely shorter than half the height of the graph (the graph has a height of approximately 44dB). That is because the formula *doesn't* calculate the length of the green line. It calculates the length of a line that is perpendicular to the red line, assuming that one x unit takes up as much space as one y unit.

The graph has approximately 44 y units from top to bottom. And 30000000 x units from left to right (416 MHz - 386 MHz). Granted, the graph is also displayed wider than it is high. But still, "one Hz" takes up considerably less space than "one dB". If we want to display the graph with equal spacing for y and x units, we need to zoom in horizontally. We need to zoom in so much that it shows a frequency range of approximately 88 Hz (double the dB range, assuming the graph is twice as wide as it is high). That would make the red line almost horizontal, and if we now calculate the perpendicular distance, the resulting line would be almost vertical. That is what the formula calculates and if you eyeball a vertical line below marker 3, it intersects the red line approximately 22 dB below the marker. It just doesn't look perpendicular because the added distortion of the graph.

Well, this looks like your formula does pretty much the same as taking the vertical distance directly, doesn't it? No, in some extreme cases (e.g. for points that are much closer in frequency[Hz] than they are in amplitude[dB]) it would more or less give the same result as the *horizontal* distance.
And all that assumes that you somehow are allowed to add values with different units, which you definitely can not.

YuryW, 2022-12-14:
I used slightly different initial data:
x1 = 3956
y1 = -38.41
x2 = 4070
y2 = -4.265
x0 = 4002
y0 = -2.166.
Сalculated
"distance" = 21.52   (perpendicular green)
D = 22.466  (vertical from marker 3 to the red line).
That is, the formula for D seems to be correct, and then "distance" is lying.
It seems to me that the approach itself is correct, since D is correctly calculated (it means that it is possible to count with the graph data), but this is a vertical distance, and it does not suit us, and there is an error in the "distance" formula. It is not surprising that the above manual talks about marker calculations, as I understand it, already with graph data, where there are some numbers for x and y and it is necessary to perform geometric calculations between them.I will look for where the error is in "distance".

I wonder if your math can calculate from the above data the correct distance from marker 3 to the red line along the green perpendicular?

jankae, 2022-12-14:
> I used slightly different initial data: [...] "distance" = 21.52 (perpendicular green)

That is to be expected. By dividing all frequency values by 100000, you essentially zoomed in horizontally (and shifted everything to lower frequencies, but the absolute values are not relevant here). This makes the calculated green line (important: that is *not* the green line I added to the graph) slightly less vertical and thus shorter.

> D = 22.466 (vertical from marker 3 to the red line).

The D formula is just calculating the vertical distance. This is exactly what I am using in my current implementation (not exactly written like that but the result is the same).

> there is an error in the "distance" formula

There is no error in the formula, it correctly calculates the distance of a point from a line. It is just not applicable here.

> manual talks about marker calculations, as I understand it, already with graph data

This is likely just the explanation for the user. I can almost assure you that no one is going to to calculations based on rendered data when you have the raw data already available.

> I wonder if your math can calculate from the above data the correct distance from marker 3 to the red line along the green perpendicular?

I can give you the length of the line in pixels and if I pretend that the line would be vertical, I could also convert these number of pixels to dB.

One last attempt to explain the problem:

All these geometric distance calculations work on graphs with the same unit on y and x axes. For example your typical map. Distances pretty obviously work on maps.

Please try to remove the rendering (the graph) from the discussion. We must at least be able to agree that any calculated value from the raw data must not change depending on the way that information is rendered at the end. Showing the data in a different way does not actually change the non-uniformity/flatness. If we can not agree on that, this whole discussion seems pointless.
Any "perpendicularity" depends on that rendering. You can easily take any perpendicular lines, and render them on a graph. When you then stretch that graph in one dimension, these lines will not look perpendicular anymore (unless they were exactly vertical and horizontal).
What all these geometric formulas assume is that one x unit takes up the same amount of space as one y unit. Otherwise something like a circle would show up as an ellipse. This is not the case with frequency/magnitude graphs.
And even worse, the axes have different units here. A definition of a "distance" simply doesn't make any sense (unless you are talking *rendered* distance but as mentioned earlier the rendering shouldn't even be considered).

Think of a graph showing height vs. age of a person. Pick two random points on the graph, draw a straight line between them. What length is that line? Sure, you can give me a physical distance if you draw the thing on a piece of paper. But in terms of height or age, how long is that line?

YuryW, 2022-12-15:
> What all these geometric formulas assume is that one x unit takes up the same amount of space as one y unit. Otherwise something like a circle would show up as an ellipse.

Then the last question in this discussion. Can we first "normalize" our graph data before calculating the very first formula to get a kind of circle, not an ellipse? That is, the x-axis is somehow brought in line with the y-axis. Indeed, at some scale in frequency (stretching or squeezing, multiplying or dividing the frequency by some factor), can we make our ellipse a circle?
Indeed, in the VNA described above, flatness is somehow measured.

dna35, 2022-12-15:
Dear Yury and Jan,

That was an incredible math journey. I can't help but interject the following hypothesis.

In the limit as dy/dx tends to zero, is the earth flat?

Thankyou for your serious consideration.

YuryW, 2022-12-15:
> Thankyou for your serious consideration.

All to ensure that the LibreVNA functionality is as good as other VNAs... The reasons why the dynamic range deteriorates at high frequencies have yet to be found. There is also a lot of interesting and incredible ....))))

jankae, 2022-12-15:
> In the limit as dy/dx tends to zero, is the earth flat?

Although this isn't entirely a serious consideration: it depends on the coordinate system ;) On a cartesian graph it is flat, on a polar graph it is round :D

> So you can safely add a similar function to LibreVNA

Okay, I will go ahead with the function (probably naming it flatness instead of non-uniformity as in VNA TR1300).

YuryW, 2022-12-16:
Once again about the formula from the first post
D = y0 - ( (x0 - x1) * (y2 - y1) / (x2 - x1) + y1) = y0 - YL0,
where
YL0 = (x0 – x1) * (y2 – y1) / (x2 – x1) + y1 (straight line formula between markers 1 and 2) .

The measured value does not depend on the dimension of the substituted frequency (Hz, MHz, GHz), because the frequency is a multiplier in both the numerator and the denominator. And so the dimension of the final value will be dB.
I calculated D for the example using the formula, and then by scaling the graph along y, compared this value with the geometric solution after each scaling - lowering the vertical to the red straight line between markers 1 and 2. During the screenshots, the cursor icon is not visible in the pictures, but it was set as as close as possible to the center of the red line at 2.8 GHz.
Initial data for calculation by the formula:
x1 = 1500, y1 = -6.599,
x2 = 4000, y2 = -25.33,
x0 = 2800, y0 = -9.567.
According to the formula D = 6.77212 dB.
[image]
[image]

Dgr1 = -9.567-(-16.18) = 6.613 dB.
[image]

Dgr2 = -9.567-(-16.29) = 6.723 dB.
[image]

Dgr3 = -9.567-(-16.35) = 6.783 dB.
[image]

Dgr4 = -9.567-(-16.371) = 6.804 dB. (Perhaps, with increasing scale, the error is associated with the thickness of the red line, which is located strongly vertically).

jankae, 2022-12-16:
> The measured value does not depend on the dimension of the substituted frequency (Hz, MHz, GHz), because the frequency is a multiplier in both the numerator and the denominator. And so the dimension of the final value will be dB.

Absolutely correct :)

> It seems that in the new edition of the LibreVNA application, markers that are not visible on the chart are no longer displayed next to the chart.

Yes, only visible markers are shown next to the graph.

I have renamed the marker type to "Flatness" and also added some basic lines to help the user understand how the measurement is made:
[image]

YuryW, 2022-12-16:
> I have renamed the marker type to "Flatness" and also added some basic lines to help the user understand how the measurement is made:

Thank you very much Jankae. Only Flatness itself as the sum of the maximum deviation in "+" and "-" is not always important. And when setting up the equipment, +dev max and -dev max deviation is important, as most manufacturers state "ripple: +-0.75 dB" for their equipment, not "ripple 1.5 dB". And this point is important - how symmetrical the deviation from the straight line is. That is, with "flatness: 1.5 dB" and +dev max = 0.5 dB and -dev max = 1 dB there will be a marriage, and for "flatness: 1.5 dB" and +dev max = 0.75 dB and -dev max = 0.75 dB it will be normal .
So that specifying +dev max and -dev max is ultimately mandatory and paramount.
That is why in VNA TR1300/1, only +dev and -dev are displayed as a result.
        
Yes, plotting -dev max and +dev max would be cool when setting up the hardware. Points where correction is needed are immediately visible. 
And of course, as shown in the figure (it is clear that for illustration) no one will use this function, but for the top of the bandpass filter they will definitely use . )

jankae, 2022-12-16:
> So that specifying +dev max and -dev max is ultimately mandatory and paramount.

Already possible (right-click the marker in the marker list, select displayed format)

YuryW, 2022-12-16:
> Already possible (right-click the marker in the marker list, select displayed format)

Where is the commit on github.com?

jankae, 2022-12-16:
Sorry, I forgot to push

YuryW, 2022-12-16:
> Already possible (right-click the marker in the marker list, select displayed format)

Thank you very much Jankae.
No, it needs some tweaking: either "flatness" or "dev+" and "dev-" at the same time. It makes no sense to indicate "dev+" and "dev-" separately - it's very inconvenient. It is better to show "dev+" and "dev-" or all three indicators in one line at once.
When setting up, you need to see the whole picture in numbers at once, otherwise it will take a very long time to switch between types of flatness.

YuryW, 2022-12-17:
When trying to measure the flatness of the Murata bandpass filter (touchstone s2p import, like Jankae above), when assigning a Flatness type to a marker, the marker runs away from the graph, so it is not shown next to the graph. But after assigning the necessary extreme frequencies in the Marker menu, then everything works normally. Why initially does the marker change frequency arbitrarily outside of the graph?

[image]

[image]

dna35, 2022-12-17:
Hi Yury,

Flatness results question.

I see the same phenomena with the cursor, but I want to address how the results are used.

Case: 1
Murata Saw Filter RF1419D
What parameters from the spec does one use to determine adequate flatness or I suppose AKA, Band Pass ripple.
I see the cursors appear to arbitrarily placed at the halfway point of the ripple.

Feel free to correct any misassumptions.

[rf1419d.pdf](https://github.com/jankae/LibreVNA/files/10252803/rf1419d.pdf)
File 1 - RF1419D Specs

jankae, 2022-12-17:
> And the middle marker could perhaps be labeled as medium "m"

The middle marker can't have a name, sorry. Only "helpermarkers" (the extra markers that appear for certain marker types) can have names. The frequency of the flatness marker is actually not important, I just chose to place it in the middle.

> Why initially does the marker change frequency arbitrarily outside of the graph?

Every new marker (or helpermarker) initially has a default position of 1GHz. If the trace doesn't have data at 1GHz, the marker is instead moved to the closest trace point (the end of the trace in this case). I agree that this could use some improvement.

> When switching between de-embedding and without it, markers jump to new characteristics automatically, and in order for the line to rebuild between markers, you need to move them at least a little.

Thanks, I'll see if I can get the line updated automatically.

> What parameters from the spec does one use to determine adequate flatness or I suppose AKA, Band Pass ripple?

I'd suggest the parameter "Passband ripple" ;)
[image]

YuryW, 2022-12-17:
> I see the cursors appear to be arbitrarily placed at the halfway point of the ripple.

Well, yes, you are right. The cursors are placed randomly, since LibreVNA was tested on the first transfer characteristic that came across. To test this filter according to "Passband Ripple (Relative to ILMIN) 402-405 MHz", I understand that in the LibreVNA application, you need to set the cursors at the specified frequencies on the transfer characteristic and measure exactly Flatness. LibreVNA will mark the ILMIN point as the top of the delat+ max segment. By moving the cursor to the delta + max top, at the same time, you can approximately see one more parameter from the datasheet - ILmin or measure it exactly using the search for the maximum with a marker in the band in LibreVNA.
There are still ambiguities with readings (frequency discreteness). You may need to configure LbreVNA Preferences > Marker > Snup to measurement points. I noticed a slight conflict between the frequency resolution in this touchstone s2p and trying to set the marker frequency in LibreVNA with a higher resolution at the above default setting. But there hasn't been time for more detailed measurement tests yet.

dna35, 2022-12-17:
Thanks Yury, great addition to functional testing. I look forward to your dissection of the new test.

YuryW, 2022-12-18:
Look, almost all amplifier parameters can be measured directly or indirectly using LibreVNA. CTB, CSO and 1dB Compressed have not yet been tested.
[image]

YuryW, 2022-12-18:
> The frequency of the flatness marker is actually not important, I just chose to place it in the middle.

Hi Jankае.
Is it possible to output not the average frequency, which does not matter to flatness, but the delta frequency, as the difference in the frequencies of the extreme markers? In this case, it will be clear in which frequency band the flatness is measured and everything will be logically connected together.
Best regards.

jankae, 2022-12-18:
As you have noticed, I am using a hidden marker at the mean of the upper and lower frequency limit. This marker is the "flatness" type marker and creates the two helpermarkers for the frequency range. So far, I am not completely happy with this solution. Maybe I will "unhide" that marker, use it as the lower frequency limit and use only one additional helpermarker for the upper frequency limit - not sure yet.

In any case, the available output formats will likely change in the future (e.g. show +delta and -delta at the same time).

YuryW, 2022-12-19:
Thank you so much.

YuryW, 2022-12-22:
> In any case, the available output formats will likely change in the future

Hi Jankae,
It would also be convenient, probably, to be able to turn off additional explanatory lines for flatness on the charts so that there are no extra lines on the traces. How flatness is measured - additional explanatory lines - it can be interesting only at first (not used to it yet).
And yet, if there are more than one flatness measurement markers, then it is desirable to prohibit other markers on the charts, when they are located next to the chart, to be placed between the flatness markers.
Best regards.

dna35, 2022-12-23:
Hi Yury,

I don't know if this is a bug or just requires an explanation. 
The Flatness varies between the Linear and Log sweeps, apparently due to the line dropping down at ~ 900 MHz.

[image]
Fig. 1 Flatness On - Linear Frequency Sweep ~1.50 dB

[image]
Fig. 2 Flatness On - Log Frequency Sweep ~ 2.9db

YuryW, 2022-12-23:
Hi dna35,
It seems to me that it is necessary to indicate in the instructions for LibreVNA that flatness measurements do not work with non-linear (log) axes on charts and when logarithmic sweep is enabled.
At the same time, you probably need to draw Jankae's attention to the fact that when you check the box on the Logarithmic sweep, the linear frequency remains on the chart at the moment, which, probably, needs to be manually switched to the Logarithmic for the time being.
And at the same time, probably, make the "0" (Zero span) button inactive in VNA mode.
Best regards.

jankae, 2022-12-23:
> The Flatness varies between the Linear and Log sweeps, apparently due to the line dropping down at ~ 900 MHz.

For this and all future observations like this: please provide the raw data (e.g. exported as touchstone). Screenshots are nice as an addition as well, but I can't replicate data dependant behavior with them.

> flatness measurements do not work with non-linear (log) axes on charts

The explanatory lines do not work. They would need to be "log"-shaped and I simply draw them as a straight line between the two markers.

The calculation itself should not be affected. During the calculation of flatness, the code doesn't even know whether the graph displays linear or log frequency (calculation always works on raw data, never on plotted data). In fact, the flatness marker may show up on two graphs at the same time, one with linear frequency and one with logarithmic frequency. 

> and when logarithmic sweep is enabled.

This should have no effect on the flatness calculation. Yes, you might get slightly different results because the measured frequency points are different ones in a logarithmic sweep. But overall, it shouldn't be too much of a difference.

> At the same time, you probably need to draw Jankae's attention to the fact that when you check the box on the Logarithmic sweep, the linear frequency remains on the chart at the moment, which, probably, needs to be manually switched to the Logarithmic for the time being.

I don't plan to change this. The graph system is independant from the sweep settings.

> And at the same time, probably, make the "0" (Zero span) button inactive in VNA mode.

Why? Zerospan mode is a possible mode of operation for a VNA. 

By the way, these last two points really feel like totally different issues to me. Please create a new issue if the discussion deviates from the original topic. Otherwise keeping track of and closing issues becomes a nightmare.

YuryW, 2022-12-23:
> Why? Zerospan mode is a possible mode of operation for a VNA.

Hi Jankae,
Exactly, I got excited ... I completely forgot about the power scan.

> Please create a new issue if the discussion deviates from the original topic.

Sorry, this won't happen again...

dna35, 2022-12-23:
> For this and all future observations like this: please provide the raw data (e.g. exported as touchstone). Screenshots are nice as an addition as well, but I can't replicate data dependant behavior with them.

Unfortunately, I can't replicate the previous setup data. 
> The calculation itself should not be affected. During the calculation of flatness, the code doesn't even know whether the graph displays linear or log frequency (calculation always works on raw data, never on plotted data).

You're right, it would have been nice to have the previous Touchstone files!

Here's the new data with the Touchstone files.

[image]
Fig. 1 Linear Sweep

[image]
Fif. 2 Log Sweep

[Flatness Weinshel 1515 Log Sweep II.txt](https://github.com/jankae/LibreVNA/files/10297561/Flatness.Weinshel.1515.Log.Sweep.II.txt)
[Flatness Weinshel 1515 Linear sweep II.txt](https://github.com/jankae/LibreVNA/files/10297562/Flatness.Weinshel.1515.Linear.sweep.II.txt)

YuryW, 2022-12-23:
> I simply draw them as a straight line between the two markers.

Vertical explanatory lines in this non-linear (log) mode do not fulfill their function. That is, the question arises again about the possibility of excluding the explanatory lines themselves from the chart in the settings.

YuryW, 2022-12-23:
> Fig. 2 Log Sweep

dna35. 
A logarithmic sweep with a linear scale in X gives only an increase in scan points in the low frequency region and rare readings at high frequency. I'm wondering - do you really need it in measurements sometimes or in this case just for a test?

dna35, 2022-12-23:
> or in this case just for a test?

Just for a test, 

> A logarithmic sweep with a linear scale in X gives only an increase in scan points in the low frequency region and rare readings at high frequency.

otherwise I'm not sure how to use it, although I do see what it does as you indicated.

YuryW, 2022-12-23:
[image]

[image]

dna35, 2022-12-23:
I see, now it makes sense. Thank you very much.

YuryW, 2022-12-25:
Hi Jankae,  
Alternatively, instead of vertical explanatory lines in the Flatness mode on the Sxx trace of the characteristic, you can indicate with additional markers the points of maximum deviation from the straight line connecting the markers of the upper and lower measurement limits in frequency. In this case, the information content of the measurement process itself will not suffer much, but it will be possible to refuse, at least in the logarithmic mode, the vertical explanatory lines on the Sxx graphs.
Best regards.

closed 2023-10-23 by jankae