jankae/LibreVNA #172: GUI Screen Size Enlarges past full Win10 screen depending on monitor resolution

dna35, 2022-12-12:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-fa8f49086
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

I haven't seen the 3 click issue recently since it has been archived, however I would like to document the LibreVNA GUI that increases in size past the full screen Win10 window which then cuts off the LibreVNA GUI on the bottom. I could not duplicate the right side going off screen at this time that I earlier attributed to excessively long filenames for the "Setup:" and "Calibration:" files at the bottom right of the LibreVNA GUI.

Documentation for change in screen size:

1- For instance, with a completely full screen of graphs and the GUI Device Log and Marker window activated, there is an opportunity for the LibreVNA GUI to extend past the bottom of the full screen Win10 window.

[image]

2- If the LibreVNA  Dock: Device and Marker windows are opened at the same time, the bottom of the GUI is cut off.

[image]

However, as I just found out It appears to be related to the Monitor's resolution.

1- Two of three monitors from left to right in order #2...#3...#1 have different resolutions.

Monitor #2 1920 x 1080 (Recommended) 
Monitor #3 1920 x 1200 (Recommended)
Monitor #1 1920 x 1080 (Recommended)

The LibreVNA GUI fits Monitor #3 at 1920 x1200 but is cut off on the bottom of Monitors #2 and #1 at 1920 x 1080. Resolutions other than the recommended ones are not resolution acceptable.

This may be an understanding of the solution, but to close or keep the bug open may depend on whether there's a reasonable chance of addressing it.

### Expected behavior

It is expected that the LibreVNA GUI should not extend past the Win10 full window mode. Doing so causes a hardship in reading the LibreVNA units connected, temperatures, Setup file, Calibration file, and taking a screen snip. Mitigation is by exiting from full screen and moving the Win10 window for acceptable viewing or moving the LibreVNA GUI to a monitor that does not cut it off.

### Extra information & Setup and Calibration files

This Page Left Blank

jankae, 2022-12-12:
Looks like the plots take up too much space when you have four rows and not enough vertical resolution. And the minimum size of the individual plots is limited, so the window simply refuses to be smaller than the minimum required size (even if this is higher than the monitor resolution).
fb347c37ac adds scroll bars to the graph areas. With that solution my window with four rows of plots fits on a screen with 1080 vertical resolution:
[image]

YuryW, 2022-12-12:
Hi.
Thank you for persistence and intuition dna35. In LibreVNA Version (64 bit): 1.5.0-alpha.2-fb347c37a
the problem of 3 clicks to launch the application in full screen is gone. Now on my monitor with 
a resolution of 3840 x 2160, the LibreVNA application starts immediately in full screen!!!
Many thanks to dna35 and jankae.
Best regards.

dna35, 2022-12-12:
You're welcome, Yury. Thankyou Jan, unfortunately I can't get build #629 to work that previously worked or #633.

dna35, 2022-12-12:
Actually, it stops working at build #628.

dna35, 2022-12-13:
OMG, I had to reboot the PC for a MS update. Latest version now loads.

LibreVNA Version (64 bit): 1.5.0-alpha.2-8778d8257
OS: Windows 10 Version 2009
CPU Arch: x86_64

When in doubt reboot still applies!

jankae, 2022-12-13:
> Even though the issue appears to be resolved, it still remains as to why the new folder was necessary.

I am not sure if this applies but you should always replace the complete folder when downloading a new compiled version. Just replacing the .exe might not be enough every time

> The Device log was empty or not scrolling by any information. Reloading the FW resolved issue.

Most likely unrelated. I have had it disappear several times but whenever I try to solve that, it pops back up and refuses to disappear.

dna35, 2022-12-13:
> I am not sure if this applies but you should always replace the complete folder when downloading a new compiled version. Just replacing the .exe might not be enough every time

The whole folder is replaced so, it remains to be seen. I did continue to have crashes working in TDR, since that was my main mode of interest but it's undocumented. May have been related to the slow scanning or de-embedding toggle but needs revisiting as necessary. Although, occasionally with prior commits I could crash the Dual VNA mode but it's undocumented.

> Most likely unrelated. I have had it disappear several times but whenever I try to solve that, it pops back up and refuses to disappear.

I'm glad that you've seen that before! That's one bug that has gotten around but it's the first time for me.

jankae, 2022-12-14:
Closing as original issue appears to be fixed

closed 2022-12-14 by jankae