jankae/LibreVNA #173: TDR Impedance Math (off) then (on) returns to Impulse Response

dna35, 2022-12-13:
### Summary

The preference would be to return to the previous Math function of Impedance. Not doing so, requires opening the Primary Y axis Setup and selecting Impedance from the pull-down menu.

1- Original TDR Math mode
[image]
Fig.1 

2-Turn off Math goes to Frequency Domain Magnitude. Okay, unless previous desirable.
[image]
Fig. 2

3- TDR Math on goes to Impulse Response - Definitely want to return to Impedance or last setting of Fig.1.
[image]
Fig. 3

4- Return to TDR Impedance instead of Impulse Response
[image]
Fig. 4

### Paths to implement it

Set the Math switch to retain the last setting in the Time Domain or Frequency Domain mode.

### Extra information

None

jankae, 2022-12-13:
Unfortunately, these are completely different parts of the software. The trace (and the math switch) only determines the output domain of the trace, the graph determines the axes configuration.

When you turn off the math, the trace output data changes from time domain to frequency domain. The graph notices that change and attempts to configure itself to match the new domain. The same thing happens when you turn the TDR back on.
The graph has no memory of its last configuration, so it defaults to some standard settings for the Y axes. I suppose it is possible to store the previous configuration in the graph as well but I am not sure yet if I will go down that route.
For now, you can just change the graph behavior in the preferences:
[image]
This will prevent the graph from changing any axes configuration when the trace data domain changes but it will instead remove the trace from the graph and you need to manually add it back. Of course, adding the trace back will require you to set up the graph for the correct domain. You could use two graphs, one set up for frequency and one for time, that way you will always have one that is configured correctly for the trace.

dna35, 2022-12-14:
Jan, I have tried the "Remove changed trace from all unsupported graphs" and have applied it to get a better understanding.

Thank you for bringing that to my attention.

closed 2022-12-14 by dna35