jankae/LibreVNA #174: Build #626 and below appears to be more useful in Dual VNA mode

dna35, 2022-12-13:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-b67275831
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

This is more of a milestone checkpoint or feasibility test report.

Even though the scroll bar is an excellent needed feature, build #626 is more useful and stable. Maybe it can be installed right after build #626. 

The GUI slows down in Dual VNA mode after build#626 with the introduction of the Toggle switch. The Toggle for de-embedding traces is very nice as long as the graph scans and stability are not affected. At this point I have rolled back to the more stable build#626.

I realize this is a WIP effort and hope that the new features will materialize. However, robust functionality is preferable.

### Expected behavior

All graphs in the frequency domain must function in a reasonable manner and at a reasonable speed. The TDR functionality must not be compromised and may actually be considered a stress test for useability or the Gold Standard.

### Extra information & Setup and Calibration files

None

jankae, 2022-12-13:
Looks like this was caused by unnecessarily emitting a signal for every added de-embedding point. That signal caused the trace widget to be updated all the time (even when nothing relevant changed). I have now changed the code to only emit the signal when the first de-embedding point is added and my sweep speed is now unchanged when I enable de-embedding with two VNAs connected.

dna35, 2022-12-13:
Fantastic, it is now working very well!

closed 2022-12-13 by jankae