jankae/LibreVNA #175: GUI Startup - Setup file apparently switches to last used GUI folder

dna35, 2022-12-13:
### LibreVNA Version

Commit #626
LibreVNA Version (64 bit): 1.5.0-alpha.2-b67275831
OS: Windows 10 Version 2009
CPU Arch: x86_64

Commit #637
LibreVNA Version (64 bit): 1.5.0-alpha.2-2c62fe71a
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

Scenario: 

The LibreVNA autosave .setup file appears to follow the last VNA setting regardless of its folder.

Folder VNAII for Commit #626

LibreVNAII GUI - Shortcut
[image]
Fig.1 

After short cut for Commit #637, the auto save switched to the VNAIIa GUI folder
[image]
Fig. 1a 

Folder VNAIIa for Commit #637

LibreVNAIIa GUI - Shortcut
[image]
Fig. 2

Conversely, the VNAIIa will switch back to the VNAII auto save folder.

I'm going to stop here because surely, I must be doing something wrong.

### Expected behavior

The autosave last setup files should be independent of other setup folders. However, after setting the autosave and then activating the wanted GUI folder, the autosave defaults to the last used GUI. This may cause the no GUI syndrome when a commit is updated.

### Extra information & Setup and Calibration files

None

jankae, 2022-12-13:
This is intended behavior. The default setup file is stored in the preferences (in the registry on Windows). As with all other preferences, there is only one global instance, regardless of how many folders with different versions of the GUI you use.

The GUI should be able to handle setup files from other versions. If you believe that the default setup file causes problems, please try the following:
1. Move the default setup file to a different location
2. If the GUI now starts, please send me the problematic setup file

dna35, 2022-12-13:
> This is intended behavior. The default setup file is stored in the preferences (in the registry on Windows). As with all other preferences, there is only one global instance, regardless of how many folders with different versions of the GUI you use.

How does one run different GUI's with different last save setup files?

jankae, 2022-12-13:
Sorry, that is not supported. Doing that would require saving different preferences for each version which I don't think is very user friendly (you would need to customize the preferences whenever you downloaded a new version). If you want different setup files, I suggest that you do not use a default setup and instead load the file manually after starting the GUI.

dna35, 2022-12-13:
Thanks Jan, I can work around this now.

closed 2022-12-13 by jankae