jankae/LibreVNA #177: Autosave Setup file works but All other setup loading files do not

dna35, 2022-12-14:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-4568be8ff
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

Introduction:

Not sure if this is a bug, but this may have started after deleting the Win10 LibreVNA registry and trying to cleanup loading issues.

In any case, the autosave file VNAII.setup seems to be the only file that loads. All other setups result in no GUI shown. 

[4 Port Investigations TXT.txt](https://github.com/jankae/LibreVNA/files/10231479/4.Port.Investigations.TXT.txt)
Appended .txt for Github but natively won't load on LibreVNA

[Auto save - VNAII-setup load okay.txt](https://github.com/jankae/LibreVNA/files/10231490/Auto.save.-.VNAII-setup.load.okay.txt)
Appended .txt for Github but natively will load on LibreVNA.

The setup is meant to be viewed in Dual VVNA mode using SOLT_1234.

### Expected behavior

Save the current setup and load it again.

### Extra information & Setup and Calibration files

None

jankae, 2022-12-14:
I can load both files just fine the first time around. But if I try to load the other file after this (or the same file again), the GUI crashes. This is likely what happens for you as well: the autosave setup gets loaded at startup and afterwards the second setup load file crashes the GUI.
I will try to debug this now.

jankae, 2022-12-14:
Uh, this was actually a beginners mistake... I forgot to make certain destructors virtual. That leads to undefined behavior in [certain cases](https://wiki.sei.cmu.edu/confluence/display/cplusplus/OOP52-CPP.+Do+not+delete+a+polymorphic+object+without+a+virtual+destructor). And unfortunately, deleting the de-embedding setup (prior to loading a new setup file) was just such a case.

Sorry, should be fixed now.

dna35, 2022-12-14:
No problem Jan, I thought the phantom of all phantoms was upon us. You're still a genius!

On another note, when saving a dual vna setup, the setup will not populate the graphs in 4 port mode until first opening one of the vna's in two port mode and then going back to the Dula VNA  mode.

Would you like me to open another Github issue?

dna35, 2022-12-15:
Thankyou Jan, I will close this and open a new issue request.

closed 2022-12-15 by dna35