jankae/LibreVNA #178: 4 port impedance vs 2 port impedance value

dna35, 2022-12-15:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-fe340ac62
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

Introduction:

I will start to address this from observable graphs but also possibly in another post with respect to the calibration files of the two VNA'a. 

First, the observable data:

Note: The impedance will show as 50 Ohms per line for single ended S parameters.

[image]
Fig.1 Anritsu 127.0 mm 100 Ohm Differential line

[image]
Fig. 2 VNA1 S11 S22 shows the same as 4 Port setup

[image]
Fig. 3 VNA2 S_11 S_22 are now inverted. 4 port mode would be S_33 S_44.

So, what happened? Is there a calibration issue between the two VNA's or a bug. Which brings one to the 2nd thoughtful question.
1- Is the calibration file for VNA1 & VNA2 separate?
2- If so, do they need to use different names?

### Expected behavior

The TDR impedance response should be the same in Dual VNA mode as individual VNA mode.

### Extra information & Setup and Calibration files

DUT setup:
[image]

Setup File:
[4 Port Investigations 4 vs 2 port.txt](https://github.com/jankae/LibreVNA/files/10233013/4.Port.Investigations.4.vs.2.port.txt)

jankae, 2022-12-15:
> The TDR impedance response should be the same in Dual VNA mode as individual VNA mode.

Yes, your result looks indeed weird.

> 1- Is the calibration file for VNA1 & VNA2 separate?

They are two different VNAs, so they need their own calibration each. Whenever you connect to a compound device (multiple VNAs at the same time), you can just consider this to be a new, larger VNA. That one needs its own calibration as well.

You should never apply a calibration file that was created on one VNA to a different VNA. This could be an explanation for the result you are seeing.

> 2- If so, do they need to use different names?

The name of the calibration is not important. What is important is that you don't mix calibrations between devices.

dna35, 2022-12-15:
> The name of the calibration is not important. What is important is that you don't mix calibrations between devices.

This is the one I'm confused about. If I separately calibrate VNA 1 with calibration file name VNA_Cal and then calibrate VNA 2 as VNA_CAL isn't VNA_CAL now the errant Cal file for VNA 1?
Should I initially be using two different VNA folders?

So then, it seemed that doing a full 4 port compound device Cal would provide all the calibration that would be needed.

Therefore, if the compound device is given the same file name, then I would lose the individual calibrations but have a set of useful dual calibrations for the compound and individual devices with the same name VNA_Cal. I believe this is the scenario that I took. Where have I gone wrong?

jankae, 2022-12-15:
> If I separately calibrate VNA 1 with calibration file name VNA_Cal and then calibrate VNA 2 as VNA_CAL isn't VNA_CAL now the errant Cal file for VNA 1?

I just meant that the GUI doesn't care about the filename or location. If you go ahead and save another calibration in the same folder with the same name, then yes, of course you will lose the previous calibration. This works just like any other file in any other program: if you overwrite it, you lose the previous content.

> Should I initially be using two different VNA folders?

Different folder and same names or same folder and different names it doesn't matter, just don't mix up your calibration files.

> So then, it seemed that doing a full 4 port compound device Cal would provide all the calibration that would be needed.

For that specific 4 port VNA, yes. You can't use that calibration for any other device.

> Where have I gone wrong?

You can not use the calibration you created with the four port VNA for a single VNA (even if it was part of the four port VNA).
There is one exception: if you have assigned port 1 and 2 of the compound device to the same device, you could use your four port calibration for that single device as well (because ports 1 and 2 of the calibration lines up with the physical ports on both the single device and the four port VNA). The same is not true for the second device in the four port VNA.
But to be honest: don't do that. It just creates further confusion.

Always use a dedicated calibration file for every VNA and consider any combined VNA as a completely new VNA with yet a different calibration file.

dna35, 2022-12-15:
To recap:

If there is a different Cal file name for the three different confurations of VNA1, VNA2, VNADual, then selecting a VNA... device from the single instance of the GUI would require manually loading the correct file each time per the two real and one virtual machine. Once again, please correct my logic as necessary.

jankae, 2022-12-15:
Yes, either manually load the correct calibration file or assign default calibration files to the devices

dna35, 2022-12-15:
Yes Jan, even though I haven't done it yet, I believe the resolution is assigning the default but differently named cal file per unit.

dna35, 2022-12-15:
It works now Jan. Early on I was saving the 2nd VNA to a unique Cal file because I was running two different instances of the GUI.
When I learned that only one instance was needed, the scheme was not perpetuated or fell apart.

[image]
Fig.1 Final Initial 4 port results

[image]
Fig. 2 Final 2 port VNA2 results

[image]
Fig. 3 Final 2 port VNA1 results

[image]
Fig. 4 Final 4 port round robin results

dna35, 2022-12-15:
> There is one exception: if you have assigned port 1 and 2 of the compound device to the same device, you could use your four port calibration for that single device as well (because ports 1 and 2 of the calibration lines up with the physical ports on both the single device and the four port VNA).

Evidently this exception applies in my case!

jankae, 2022-12-15:
> I believe the resolution is assigning the default but differently named cal file per unit.

Yes, that would be a good solution.

> However, setup files would not have the correct unique Cal file name per VNA unit. Therefore, switching between units would still not be seamless. Is there any way the GUI can help manage this?

And for precisely that reason, the calibration is not included in the setup file. You'll notice that the calibration is always reset when you load a setup file. You will have to manually load the correct calibration afterwards (unless you have a default calibration assigned to the currently connected device. In that case, this calibration will get loaded).

Also see section 2.2.3 of the manual for all the things included in the setup file.

jankae, 2022-12-15:
To prevent future confusion, I have added a warning when a calibration from another device is loaded (will only work with newer calibrations as old calibration files don't contain the correct device serial)

dna35, 2022-12-15:
I have seen the warnings previous to this commit but didn't know what they were for. (Good)

1- After, apply individual calibrations to the VNAs, the messages stopped. (Good) 
2- The caveat is the calibrations are no longer in comparable form to the DualVNA Cal, therefore making it difficult make comparisons. (Not Good)
3- Since, in the exception case where VNA1 matched the DualVNA Cal (Preferable), can't we split separate calibration files for VNA1 and VNA2 with of course unique names from the 4 port Calibration. Therefore, rendering the individual VNA calibrations redundant and unnecessary and at the same time allowing for more robust replication of DUT responses. (Preferable)

jankae, 2022-12-15:
> I have seen the warnings previous to this commit but didn't know what they were for.

No you haven't (or are referring to other warnings). I have just added them with the latest commit.

> 2- The caveat is the calibrations are no longer in comparable form to the DualVNA Cal, therefore making it difficult make comparisons. (Not Good)

I don't understand what you mean by "comparable". The warnings didn't change anything in the calculations, it is just a warning to make the user aware of a potential problem.

> can't we split separate calibration files for VNA1 and VNA2 with of course unique names from the 4 port Calibration

I am not adding that functionality. In theory, yes you can extract both individual VNA calibrations from a full 4 port calibration. But there are so many things that can go wrong here (port order, VNA order) that I think the risk of a user error is much higher than the additional effort to create a separate calibration.

For precise measurements, you should calibrate right before the measurement anyway, I really do not see a use case for transferring calibrations between devices.

dna35, 2022-12-15:
> No you haven't (or are referring to other warnings). I have just added them with the latest commit.

There was something similar about saving ones calibration file but now there are two messages. I believe it may have been only one before but never looked under the message until today. So, probably not the same message but there was a warning.

> I don't understand what you mean by "comparable". The warnings didn't change anything in the calculations, it is just a warning to make the user aware of a potential problem.

This is not for the warnings but for the DUT response. The DUT response would be the same for VNA1 in the exception case which is desirable.

> I am not adding that functionality. In theory, yes you can extract both individual VNA calibrations from a full 4 port calibration. But there are so many things that can go wrong here (port order, [...]

That is understandable but I was just asking.

> For precise measurements, you should calibrate right before the measurement anyway, I really do not see a use case for transferring calibrations between devices.
 
Yes, I will see if redoing the calibration for the individual devices will help.

dna35, 2022-12-16:
Thanks for all your help and of course a great product!

closed 2022-12-16 by dna35