jankae/LibreVNA #179: VNA 4 port setup won't show 2nd VNA graphs initially

dna35, 2022-12-15:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-fe340ac62
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

When a dual VNA setup is saved while in dual vna mode, upon loading the setup, the graphs for VNA2 will not populate.
Selecting either VNA from the Device-> Connect pulldown menu and immediately returning to the Dual VNA mode restores 2nd VNA graphs.

[image]
Fig. 1 Normal view of 4 port setup

[image]
Fig. 2 Same setup loaded now is 2 port mode. The Device-> Connect will show as Dual VNA

After switching to 2 port mode and back to Dual VNA, the graphs will be restored as in Fig. 1
1- An additional consequence is the SOLT_1234 calibration needs to be selected.

### Expected behavior

After saving a 4 port mode, it should return to the same state as when it was saved.

### Extra information & Setup and Calibration files

Setup file:
[4 Port Investigations II.txt](https://github.com/jankae/LibreVNA/files/10233191/4.Port.Investigations.II.txt)

dna35, 2022-12-16:
Jan, since saving the individual VNA cal files, the behavior here is now acceptable. VNA2 came up with a SOLT_12 calibration and one click to DualVNA with SOLT_1234 already active.

dna35, 2022-12-16:
Sorry for the confusion.
I realized the behavior is still the same if the DualVNA is saved off and subsequently loaded again. Since, it is recalled in the DualVNA mode, then yes one must go to a single device and back to the dual device to populate the VNA2 graphs.
However, it does now load with the SOLT_12343 activated.

jankae, 2022-12-16:
You are right, I could replicate the problem.

The issue was my handling of the device connection when loading setups:
- I disconnect the device first (this prevents excessive configuration of the device while deleting/creating modes)
- Configure the GUI according to the setup file
- Reconnect to the previous device:
https://github.com/jankae/LibreVNA/blob/0cc5968b12d9a044f525eb8f5afe8a29ab34a63a/Software/PC_Application/LibreVNA-GUI/appwindow.cpp#L1199

The problem is the last step: while it connects to the same device again, it does not set up some important Qt connections. As a result, the information that the device has 4 ports is not passed on properly and the VNA mode defaults to 2 ports.

Instead, I must call this function when reconnecting:
https://github.com/jankae/LibreVNA/blob/be340f2adc69983a05ba79e34316e9661121789e/Software/PC_Application/LibreVNA-GUI/appwindow.cpp#L1199

It handles the connection as well as making all the right signal connections within the GUI application. be340f2a should fix the problem.

dna35, 2022-12-16:
Nice fix! It now returns to 4 port mode after loading the saved setup.
How does one get the Autosave file to do the same thing. After exiting the GUI, it will return to the 2 port mode.

jankae, 2022-12-16:
The autosaved setup file does not include the device that is connected. The GUI always connects to the first device found when it starts. Since individual devices are always above the compound devices in the device list, it will always connect to a single device.

By the way: there is no 2 port mode or 4 port mode in the GUI. The GUI just works with what the connected device advertises. The behavior after startup is different than the behavior in the original issue:
- Original issue: connected to a compound device, but incorrectly ignoring the reported 4 ports and only using 2 -> bug
- Startup behavior: connected to a single device, correctly only using 2 ports -> not a bug

If you want to connect to a specific device after startup, you need to pass the serial number via the --device parameter when calling the GUI:
```
./LibreVNA-GUI --device <serial>
```

dna35, 2022-12-16:
Thank Jan, super fix. I understand the device hierarchy and may apply the specific device startup as recommended.

closed 2022-12-16 by dna35