jankae/LibreVNA #181: GUI disappears when setting up Custom Line Limits for Dual VNA

dna35, 2022-12-16:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-0cc5968b1
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

This is the first time using this function:

The GUI disappears when working with the Custom Line Limits. This has happened several times in TDR mode but not yet for the one time tried in frequency mode.

The Edit Line dialog almost requires adding all the horizonal rows prior to entering X & Y values. If not, then after the 2nd entry, additional row will not be appended to the last row but after the 1st row. Doing so, requires manually entering the X Y data up each row at a time. At some point in time while entering the row information in TDR mode, the GUI will disappear. I have used the mouse, enter key and Tab key for moving between cells to find the best behavior but have not settled on the best practice yet. This was prior to the VNA calibration file issues just in case there's causation.

After several crashes I was able to complete two very nice Custom Line Limit masks.

[image]
Fig. 1 

### Expected behavior

Pushing the plus (+) button should add rows at the bottom of the list. Adding X & Y information should have less anomalies.

### Extra information & Setup and Calibration files

None

jankae, 2022-12-16:
Easy way to trigger the bug from a default setup:
- Right-click a XY-plot, select axis setup
- Add a constant line (any line will do, just add at least two points)
- Close both dialogs
- Open the dialogs again, no need to edit anything
- Crash happens when closing the "Edit Line" dialog

The problem:
- Creating the "Edit line" dialog sets up a signal/slot connection to update the line description in the "Axis Setup" dialog when the "Edit line" dialog is closed
- This connection was done with a lambda without assigning a receiving object -> connections remains even when both dialogs are closed
- Opening the dialogs again creates another connection on top of the still existing connection. But the old connection tries to update the line description in the old "Axis Setup" dialog, which doesn't exist anymore -> crash

The fix (e35766c):
- Assign the correct signal receiver, which automatically disconnect the connection when the "Edit line" dialog is closed

jankae, 2022-12-16:
> Pushing the plus (+) button should add rows at the bottom of the list. 

The list of points in the line must always be in a sorted order with ascending x coordinates. Your added point will be placed in the correct row (usually at the top, because it has an x coordinate of zero when it is created). Whenever you change an x coordinate, the list will sort itself again to maintain the correct order.

dna35, 2022-12-16:
Another great fix! The coordinate row schema now seems reasonable knowing that it is auto sorting.

closed 2022-12-16 by dna35