jankae/LibreVNA #183: SA Phase Noise shows as dbm/Hz instead of dbc/Hz

dna35, 2022-12-18:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-5ad44c780
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

SA Cursor Marker on side bar shows dbm/Hz instead of dbc/Hz
The Data in Marker window shows correct at dbc/Hz.

[image]
Fig.1 Side marker note shows dbm/Hz instead of dbc/Hc

### Expected behavior

Side bar and Marker window should show the same units,.

### Extra information & Setup and Calibration files

none

YuryW, 2022-12-18:
Hello Aaron.
dBm/Hz (here dBm is the unit of measure) and dBc/Hz (the "c" stands for "carrier") are completely different quantities, unrelated. The first shows the power spectral density of the noise in a 1 Hz bandwidth with respect to 1 mW so that you can always get (calculate) the noise level versus bandwidth (RBW) of the device. The second shows the "purity" of the sinusoidal signal - how much the power in the 1 Hz bandwidth at the carrier frequency differs in dB from the power in the 1 Hz bandwidth at a certain offset from the carrier frequency. That is why only next to dBc / Hz there is always a frequency offset from the carrier.
Best regards.

dna35, 2022-12-18:
Hi Yury,

Thank you, precisely what I needed to know!

dna35, 2022-12-18:
Hi Yury,

So basically, with that understanding an on-line calculator is not converting anything but just subtracting signed numbers with the knoweldge of carrier offset.

YuryW, 2022-12-18:
Hi dna35.
dBm/Hz. A measure of power spectral density. It provides a ratio of the power in one Hertz of bandwidth, where power is expressed in units of dBm.
(https://www.mpirical.com/glossary/dbm-hz)

And dBc / Hz is the difference between the two measured values ​​dBm / Hz at the carrier frequency and at the frequency with a offset along it, which is always announced at the same time.
(https://en.wikipedia.org/wiki/DBc) (https://www.rp-photonics.com/decibel.html).
The carrier is measured as (dBm/Hz) in this case, as I understand it, only to be able to subtract the value of the same noise measurement at the frequency offset. If two carriers were being compared, then /Hz would not be needed.
dBc/Hz is the difference between two dBm/Hz. A dBc is the difference between two dBm.
Let's ask Jankae why he put the carrier level value in dBc/Hz next to the marker to confuse you.))
Best regards.

dna35, 2022-12-18:
Hi Yuri,

Magnificent, I'm glad this post will be available for reference.

**Yes, it does work out.** Albeit, there was added confusion when adding the TOI because the Marker points have to be referenced correctly but when turning off TOI Port2 it is clearer.
So, it does work out after all. Just now I had to restart the GUI because the dbm/Hz was not showing, but other than that it is now well understood.

YuryW, 2022-12-18:
> because the Marker points have to be referenced correctly

Because to measure third-order nonlinearities with two carriers, another theory is needed.

dna35, 2022-12-18:
FYI, the DUT is a CTI 100MHz PLXO Oscillator to VNA1 with a 10MHz input from VAN2. The name plate value for Phase Noise is -150dBc.

jankae, 2022-12-18:
> SA Cursor Marker on side bar shows dbm/Hz instead of dbc/Hz
The Data in Marker window shows correct at dbc/Hz.

The marker format can be configured independently for the table and the graph. You just have configured the marker on the graph to show noise. If you switch it to phase noise, you will get the correct unit of dBc/Hz.

> The name plate value for Phase Noise is -150dBc. [...] Maybe, the offset was meant to be 10kHz or even 100KHz?

Your oscillator has a much better phase noise than the LibreVNA. It is not surprising that you do not get the -150dBc value. In your setup, you are measuring the phase noise of the LibreVNA itself instead.

jankae, 2022-12-18:
I'll close this as the markers work correctly

closed 2022-12-18 by jankae