jankae/LibreVNA #185: Graph does not reset amplitude until 1st overwrite

dna35, 2022-12-22:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-77dfabc67
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

When changing a slow scanning graph, the amplitude overwrites as it scans. For a slow RBW of 100 Hz, it will look unusual.

[image]
Fig. 1 The red and blue lines are dropping down from the previous "Y" axis values.

If the red and blue are paused, it will take a full cycle of the red & blue un-paused to clear the graph of the vectors.

### Expected behavior

Reset amplitude to the starting point of trace if beginning a new trace.

### Extra information & Setup and Calibration files

None

jankae, 2022-12-22:
This issue is unclear to me. In particular, I do not understand what you mean by "reset amplitude". Does this refer to deleting/clearing old trace data? Or do you mean something related to the autoscaling of the Y-axis (amplitude)?

The trace should reset (get cleared) when:
- Sweep settings are changed
- A new device is connected
...essentially whenever new sweep settings are transferred to a device.

The trace should **not** reset (get cleared) when:
- if you pause/un-pause a trace, as this does not reconfigure the VNA. It simply ignores new incoming data for the paused trace

dna35, 2022-12-22:
> * if you pause/un-pause a trace, as this does not reconfigure the VNA. It simply ignores new incoming data for the paused trace

Yes, this would typically be an acceptable case, but when the RBW is extremely low, averaging is used, graphs are paused and restarted then the graph clutter builds up. I'm just showing what I see from a user point of view.

jankae, 2022-12-22:
In my opinion that is exactly the purpose of the pause: You can freeze data and look at it while the actual measurement might change. I understand that an argument could be made for resetting the trace when it is un-paused. However, this wouldn't get you new data faster - it would just not show the old data anymore.

But since you, the user, know that you paused the trace, you also kwow that it still contains old data. And maybe you even want to see that old data (e.g. to observe changes in the measurement). As of now, I am not convinced that un-pausing a trace should clear all prior trace data.

dna35, 2022-12-22:
I agree with your assessment, that's why making this case is difficult.

jankae, 2022-12-22:
> The point is to not have the trace drop down as a vector while it plots the current trace after the "Y" axis is changed.

Okay, I need a step by step guide for that. I still don't understand at what point you change the Y axis. If you have a "hole" in your trace (e.g. values from 1-2 GHz and from 4-5 GHz), the last point before the hole (at 2 GHz) will be connected to the first point after the hole (at 4 GHz) with a straight line. The trace doesn't really support a concept of multiple disjointed sub-traces.

You could achieve such a hole by pausing/un-pausing a slow sweeping trace. Is that what you are referring to?

dna35, 2022-12-22:
The scenario is this:

AT 100Hz RBW and no noise source attached, the ambient noise level is below -121dBM. When the noise source is attached but not activated, the noise increases slightly. When the noise is turned on after going through a 30dB amplifier, the noise increases by 10dBm. 

[image]
Fig.2 This is similar after pausing and starting

The difference between Fig.1 and Fig. 2:

Fig.2 shows pausing and starting the traces which is the typical behavior I'm aware of, but in the extreme case of Fig, 1 there must have been data from a "Y" value that differed greatly.

At the moment this may be a moot point and we can move on.

dna35, 2022-12-22:
Changing to the flat top filter is the most likely change in "Y" value but this is not a bug.

[image]
Fig. 3

dna35, 2022-12-22:
If there are no further comments, I will close this issue as a non-bug.
Thank you for your assistance, Jan!

closed 2022-12-22 by dna35