jankae/LibreVNA #186: LibreVNA Win10 Registry causing no GUI start

dna35, 2022-12-22:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-77dfabc67
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

Could not load a setup file after saving it... then the GUI would not load. Deleted the registry and it is now working again.

Prior to this, for the first time, I was comparing two setups in Word because one was not loading. 

[libre VNAregistry bu II.txt](https://github.com/jankae/LibreVNA/files/10284066/libre.VNAregistry.bu.II.txt)
File 1 Corrupted Registry

[4 Port Mag Extensions De-embedded Amplifier Testing WON'T LOAD.setup.txt](https://github.com/jankae/LibreVNA/files/10284125/4.Port.Mag.Extensions.De-embedded.Amplifier.Testing.WON.T.LOAD.setup.txt)
File 2 Setup file file that would not load

### Expected behavior

GUI should load

### Extra information & Setup and Calibration files

None

jankae, 2022-12-22:
I believe these could actually be two separate issues. I'll take care of the setup file first. The parsing didn't like the usage of de-embedded traces as the source in math traces (such as SDD11). Should be fixed now with b41b3e2

But I noticed something else in your setup: You have broken math traces. Look at SDD12 for example:
[image]
It depends on S12, S14, S32 and S34. However, only S32 and S34 are available as math sources (I did scroll all the way to the bottom, there are no more active math sources). This is not a bug in the parsing of the setup file, your file actually contains only two math sources for that trace:
[image]

Did you delete any traces after performing the mixed mode conversion?

dna35, 2022-12-22:
I cleared the math traces for SDD11 but it still crashed on loading.

jankae, 2022-12-22:
Well, yeah it will do that, there are other math traces as well. But it shouldn't crash anymore once you downloaded the fix.

dna35, 2022-12-22:
Yes, it doesn't crash now.

dna35, 2022-12-23:
> I believe these could actually be two separate issues. I'll take care of the setup file first. The parsing didn't like the usage of de-embedded traces as the source in math traces (such as SDD11). [...]

I don't know; this was early in the Mixed Parameter setups, so I would need to revisit it.

Thank you for bringing it to my attention!

dna35, 2022-12-23:
Closing as completed.

closed 2022-12-23 by dna35