jankae/LibreVNA #187: SOLT vs Through vs v1.4.0xx vs 1.5.0xx

dna35, 2022-12-25:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-3d51f1f49
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

Not sure if this is best suited here or in groups but considering the Dual VNA use I opted for here.
Is there a bug or is it user intervention? See "Extra Information & Setup" for additional files.

1- Which filter response portrays the correct shape with respect to Case 1 vs Case 2?
2- Why does the SOLT_21 response in Fig. 1 look very different than the SOLT_21 response in Fig. 2 & Fig. 3? (It actually mimics the response of the Through Calibration.)

Case 1:  v1.4.0xx AM Band Stop Filter Response
1- Photos show the SOLT (assumption) response from an AM Band Stop Filter with the setup, but no s2p files saved at that time.
[image]
Fig. 1: AM Band Stop Response using SOLT calibraton in 2021 using original VNA unit 

Case 2: v1.5.0xx AM Band Stop Filter Response

[image]
Fig. 2: AM Band Stop Response, Dual VNA using original VNA unit.  S21 through Norm_12 vs S21 SOLT_12 vs S21 Cal Off

[image]
Fig. 3: AM Band Stop Response, Dual VNA using original VNA unit expanded "Y" scale

### Expected behavior

I would expect the S21 SOLT_12 responses to be similar for Case: 1 and Case: 2.

### Extra information & Setup and Calibration files

[image]
Fig. a: AM Band Stop Filter 

[AM Band Stop Filter II.txt](https://github.com/jankae/LibreVNA/files/10300688/AM.Band.Stop.Filter.II.txt)
File a: AM Band Stop Filter setup, in 2021 with original VNA unit for Fig. 1.

[AM Band Stop Filter II.txt](https://github.com/jankae/LibreVNA/files/10300693/AM.Band.Stop.Filter.II.txt)
File b: AM Band Stop filter setup, 12/2022 with original VNA unit for Fig. 2 and Fig. 3.

[Single VNA SOLT_12 vs Through_Norm_12 II.txt](https://github.com/jankae/LibreVNA/files/10300694/Single.VNA.SOLT_12.vs.Through_Norm_12.II.txt)
File c: s2p for  VNA SOLT_12 vs Through_Norm_12 vs No Cal

YuryW, 2022-12-26:
> 1- Which filter response portrays the correct shape with respect to Case 1 vs Case 2?

Hi dna35,
I will express my opinion. LibreVNA Version (64 bit): 1.5.0-alpha.2-3d51f1f49.
I made similar measurements: I compared the measurements of the notch filter with the calibration of SOLT_12 and ThroughNormalization_12. There was a slight difference (S21, S12), but it is explained, in my opinion, by the fact that in the second case of calibration, reflected waves are not taken into account and in cases where their influence is large (when measuring negative dB), there will be measurement errors to a greater or lesser extent (depending on reflections in measuring circuit). The circuit uses 50/75 resistive transformers, so there is such a difference when measuring without calibration.
So, the first conclusion is that it is better to measure notch filters when calibrating SOLT_12. S21 measurements with different SOLT-12s may differ for several reasons: different limiting frequencies of measurements and calibrations, difference in the number of measurement points during measurements and during calibration (probably that is why these two parameters are remembered in the name of the calibration file as a reminder), measurements of large attenuations at an unheated LibreVNA device. In rare cases, there may be inconsistencies with different settings during Preferences>Marker>Positioning>"Interpolate between points" or "Snap to measurement points" measurements or calibrations.
If there is a measurement error, look first of all for errors in the measurements and calibrations themselves.
The LibreVNA application finds some errors and warns, for example, in yellow Calibration, as you have on your screen.

[image]
[image]
[image]
[image]

dna35, 2022-12-26:
Hi Yury,

Thank you for the kind in depth response. I now have enough information to continue my research. Incidentally, the Nooelec site at https://nooelec.com/store/distill-am-broadcast-am-filter.html has a Rigol measurement of the same Notch Filter that I was unaware of.

[image]
Fig.1 DC to 5 MHz sweep

dna35, 2022-12-26:
> The LibreVNA application finds some errors and warns, for example, in yellow Calibration, as you have on your screen.

Interesting, the start frequency of 400 kHz differed from the calibration start of 1 MHz which caused the yellow warning.

dna35, 2022-12-26:
I will close this as a non-bug.

closed 2022-12-26 by dna35