jankae/LibreVNA #188: SG Calibration Routine

dna35, 2022-12-27:
### Summary

LibreVNA Version (64 bit): 1.5.0-alpha.2-3d51f1f49
OS: Windows 10 Version 2009
CPU Arch: x86_64

Considering the prevalence of two LibreVna's used as SG's for common measurements, developing a calibration routine for the output magnitude would be nice regardless of phasing.

[image]
Fig. 1 VNA1 to both inputs of PicoScope with a reasonable amplitude match

[image]
Fig. 2 VNA1 & VNA2 Magnitude mismatch of ~ 10mV Pk to Pk without considering the phase missmatch

### Paths to implement it

Calibration routine based on a known dBm level.

### Extra information

None

jankae, 2022-12-27:
This already exists - see section 9.1 in the user manual (Source Calibration)

dna35, 2022-12-27:
Hmm, I'm using the same source calibration. therfore, it must need to be instrument specific.
So just to confirm, that means three different source calibrations.

1- Source Cal VNA1
2- Source Cal VNA2
3- Source Cal VNA1&2

jankae, 2022-12-27:
No, the is one source calibration per physical device. It is stored on the LibreVNA itself. But of course there are differences between samples.

So you need:
VNA1: Source calibration 1
VNA2: Source calibration 2
VNA1&2: nothing special needs to be done, they both use their respective source calibration (if one is available)

dna35, 2022-12-28:
Thanks Jan, I'm working on the Source calibrations and will close this as already implemented.

closed 2022-12-28 by dna35