jankae/LibreVNA #190: Command via SCPI: Timing & Temperature Problem

rainermueller77, 2023-01-13:
### LibreVNA Version

LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

- Load attached Setup (S21, 1 - 3 GHz, 1601 points, 1kHz)
- Control LibreVNA over SCPI commands
- Do single sweeps and read data in a loop (see command list below)
--------------------------------------------------------------------------------------------
//Configure LibreVNA for single sweeps
:DEV:MODE VNA
:VNA:SWEEP FREQUENCY
:VNA:STIM:LVL -10
:VNA:ACQ:AVG 1

//Sweep Parameter
:VNA:FREQuency:START 1000000000
:VNA:FREQuency:STOP 3000000000
:VNA:ACQ:POINTS 1601
:VNA:ACQ:IFBW 1000

//Do 2400 sweeps
for (i=0, i++, i<2400){
   // Sweep ausführen
   :VNA:ACQ:SINGLE TRUE  // start a single sweep
   Sleep(400)
   do {
      :VNA:ACQ:FIN?
      Sleep(50);
      read_data(&SocketID, ReadString);
      Sleep(50);
   } while (strncmp("TRUE", ReadString,2)==0);
  // Trace auslesen
  :VNA:TRAC:DATA?
  Sleep(50);
  read_data(&SocketID, ReadString);
}

### Expected behavior

ISSUE 1:
After command a single sweep (:VNA:ACQ:SINGLE TRUE) sometimes it can take up to 400ms until the answer of :VNA:ACQ:FIN? switches to TRUE. Typically it takes less than 100ms. It should switch to TRUE immidiatelly after sending the sweep command

ISSUE 2:
The temperature seems to become very high (until the device crashes) when doing single sweeps compared to contimous sweeping.
Using the programm below the temperature rises over 90°C within less than 2 hours (see picture). When the LibreVNA just loads the attached setup and is running in the continous sweep mode with the same parameters the temperature stays below 50°C.

### Extra information & Setup and Calibration files

Setup file and picture
[S21setup.zip](https://github.com/jankae/LibreVNA/files/10412022/S21setup.zip)
[image]

jankae, 2023-01-13:
> LibreVNA Version (64 bit): 1.4.0-7f6b78992

Can you please check with the latest version? I know that it isn't released yet but I am fairly sure that this problem

> It should switch to FALSE immidiatelly after sending the sweep command

has been fixed in the meantime. I can't find the commit right now but I do have a testcase for that exact scenario in my integration tests and it passes.

> The temperature seems to become very high (until the device crashes) when doing single sweeps compared to contimous sweeping.
Using the programm below the temperature rises over 90°C within less than 2 hours (see picture).

Have you verified the temperature by touching the device? (maybe not directly if it is actually that hot). But I am fairly certain that this is a temperature read-out bug. There is no way the PLLs are up to 95 while the MCU is still at 34. Again, I'd like you to try with the latest version and verify the bug. There have been close to 200 commits since the version you are using.

rainermueller77, 2023-01-14:
I checked with the latest version. Issue 1 is fixed! I also couldn't reproduce issue 2, so I assume it is fixed. 
But I got another strange behavior:

After a couple of 1000s sweeps, sometimes not all frequency points are measured. With the settings from above all frequency points between 1 GHz and 1.1 GHz are missing which leads to less measured points when reading via SCPI. In the LibreVNAGUI I can also see that the missing points were not measured.

I implemented a work around:
- Check the number of frequency points
- If  number is less then expected, do another sweep and read again.

closed 2023-10-23 by jankae