jankae/LibreVNA #194: Magnitude vs Frequency Graph Freezes at Midnight

dna35, 2023-02-14:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-a0fbd3ba8
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

Wait until midnight, 12 AM to watch the frequency vs magnitude graph freeze or set PC time to just before 12 AM and watch the graph freeze.

### Expected behavior

The graph should continue to refresh data normally.

### Extra information & Setup and Calibration files

The impedance vs time graph refreshes normally and is not affected.

[SOLT_12 VNA2 100kHz to 18 GHz 01-15-2023.txt](https://github.com/jankae/LibreVNA/files/10735417/SOLT_12.VNA2.100kHz.to.18.GHz.01-15-2023.txt)
File 1. SOLT Calibration 100kHz to 18GHz

[2 Port Mag VNA_2 with TDR.txt](https://github.com/jankae/LibreVNA/files/10735468/2.Port.Mag.VNA_2.with.TDR.txt)
File 2. S12 Frequency vs Magnitude & S11 TDR Port_1

[image]
Fig. 1 Normal Operation

[image]
Fig. 2 Freq vs Mag freeze and cursor erases vertical bar vs Normal TDR with 50 Ohm Term removed to show refresh

jankae, 2023-02-15:
I can confirm this, thank you very much for narrowing it down. At first glance this looks like a bug in the Qt framework (QTimer not working anymore when the time rolls over to the next day).
Depending on many settings in the graphs I either schedule a replot with a timer (e.g. for incoming new data - replotting the graph for every single new datapoint would take too much processing power) or replot immediately (e.g. when an axis settings has changed during autoscale). Directly replotting still works but the timer callback is never called after midnight. I'll see if I can come up with a solution/work-around.

jankae, 2023-02-15:
Looks like the QTimer just can't handle the transition on midnight due to QTime resetting to zero on the next day. I still believe this is a bug in Qt but I can easily work around by restarting the timer on the next day, I just have to handle the QTime overflow in my own code as well.
With 045cf6c, I no longer get a frozen graph on midnight.

dna35, 2023-02-15:
Thank you very much Jan! The integrity of the LibreVNA, in my opinion, has now been enhanced considerably.

closed 2023-02-15 by dna35