jankae/LibreVNA #195: Device Conflict connecting 2nd LibreVNA to USB

dna35, 2023-02-15:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-045cf6cc6
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

This is a known issue, but how can this be escalated to improve the user experience.

Start 1st VNA.
Start 2nd VNA

[image]
Fig.1 USB Device Error although it can be bypassed by clicking "OK".

### Expected behavior

To not click on USB warning.

### Extra information & Setup and Calibration files

None

jankae, 2023-02-15:
Yes, known issue and outside of my power: https://groups.io/g/LibreVNA-support/message/1463
This really is a problem of libusb on windows. They closed the issue I raised over there but not because they fixed the problem, they just updated the documentation to reflect the (problematic) behavior. Nothing I can do here, sorry, you'll just have to live with it.

closed 2023-02-15 by jankae

dna35, 2023-02-16:
Hi Jan,

 

Thanks for your response.

I just thought of a work around… I’ll see if I can find a Win10 “Macro App” that can help in this regard.