jankae/LibreVNA #199: SCPI command :DEVice:CONNect does not properly find and connect to available LibreVNA device

MonsieurGamgee, 2023-03-17:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-87c1f4d57
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

- LibreVNA app is launched in background
- TCP connection is made to device from host computer
- SCPI call ":DEVice:CONNect" is sent to device
- 'Device not found' is returned

### Expected behavior

As listed in the programming guide, I would expect the ":DEVice:CONNect" with no serial number given to connect to the first available LibreVNA device.

### Extra information & Setup and Calibration files

*Connecting directly to a known serial number still works* 
- Given SerNumHere is serial number of particular device:
- if SCPI call ":DEVice:CONNect SerNumHere" is sent to device
- empty char array and subsequent ":DEvice:CONNect?" call returns SerNumHere, indicating good connection

*Serial number of available device is still listed with the appropriate SCPI call*
- SCPI call ":DEVice:LIST?" returns SerNumHere

jankae, 2023-03-17:
Hi, you are absolutely correct, I broke that command in one of the recent changes. Should be fixed now with 54de9fd

MonsieurGamgee, 2023-03-17:
That did it. Thanks!