jankae/LibreVNA #201: Poor lines

Chorbu, 2023-03-19:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-54de9fd6b
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

Since a few months i have strange traces with a lot of non linearity traces. Calibration is OK,.
Il tryed to change USB cable, and try it on another PC : Same thing happened
My hamradios are off (VHF and UHF) , no spurious emissions near the VNA.
I tried to change IF1 and IF2 but without result,
Actual values are

[image]

I also put a good power supply by connect a DC power supply >2A (same result)

I hope this is not due to a degradation of components inside VNA.

[image]

### Expected behavior

What do you think about that ?

Could you give me original values of IF1 and IF2 ?

### Extra information & Setup and Calibration files

no extra information

jankae, 2023-03-19:
Hm, looks strange. Your IF values are at the default. Can you please try older firmware/GUI versions to determine whether this is a hardware or software problem? You can also enable "capture raw receiver values" in the preferences (either under Debug or LibreVNAUSBDriver, depending on your GUI version). This gives you the option to create traces from just the port or reference receiver instead of Sxy. That might get us additional insight about the source of these spikes/noise.

Chorbu, 2023-03-19:
I 've just try this software and firmware (before is not possible)
[image]

Same issue....

jankae, 2023-03-19:
Okay, I think that rules out the possibility of a new bug, I think other people would have noticed and notified me by now if that was the case.

Please look into the raw receiver values. Another thing you could try it starting the manual control dialog, enabling all PLLs and the amplifier and take a screenshot of that. On its own this is probably not that helpful but if I compare it to my LibreVNAs we might see that some values are off/different.

Chorbu, 2023-03-19:
Here is the screenshot of Manual control dialog
[image]

I activate Capture RAW receiver but i don't know where the values are saved...

Chorbu, 2023-03-19:
i forgot to activate Amplifier...

Chorbu, 2023-03-19:
But same issue

jankae, 2023-03-19:
> I activate Capture RAW receiver but i don't know where the values are saved...

You can change a trace to use the raw receiver value here:
[image]
All parameters except for RawPort1Stage1Ref and RawPort2Stage0Ref would be interesting to me.

> But same issue

Can you please post a screenshot of the manual control dialog with the amplifier enabled? The ADC values should get much larger.

Chorbu, 2023-03-19:
OK thank you

[image]

jankae, 2023-03-19:
I assume you still have something connected to the ports? Otherwise the port 2 ADC values are much too high. But otherwise that looks good. This is from my device:
[image]

Chorbu, 2023-03-19:
Here all the screens for RAW
[image]
[image]
[image]
[image]
[image]
[image]

Chorbu, 2023-03-19:
I i have a band stop filter 144MHz to 148MHz connected to my VNA

Chorbu, 2023-03-19:
So I'm going to stop it for tonight, tomorrow work and up at 6am.

Thank you in advance Jan, good night to you too.
Patrick

Chorbu, 2023-03-19:
Oh i ve just see you've already add APPLY button for my previous request !!!! Thank you for your extreme speed !!!!

Good night 
Patrick F5SUE

jankae, 2023-03-19:
You raw receiver data doesn't reveal any obvious issues (maybe a bit too much noise on the reference receiver data, but your problematic signals are coming from the port receivers). Just to make sure that it isn't from the DUT: what happens if you change the setup? (e.g. just measuring a direct through).

You can also try to reduce your IF bandwidth. Some narrow band filters have quite the long settling time and might not have fully settled in the time the VNA takes to measure a point for higher IF bandwidths. That could easily result in the measurement data you are getting.

Chorbu, 2023-03-19:
With 500Hz IF BW or more is the same issue. Tomorow evening i will try again my other filters, splitter, coupler etc...

Chorbu, 2023-03-19:
THRU connected (I SMA) after calibration

S12 is Magnitude
[image]

jankae, 2023-03-19:
Yeah, that is too much noise on the reference receiver. Same settings on my VNA:
[image]
I am unsure why you are getting that. Let's both sleep on it, maybe there are new ideas tomorrow.

Chorbu, 2023-03-19:
Yes i saw that !!! Ther is a problem with my receive port !

Chorbu, 2023-03-19:
Thank you Jan, thanks for your help. Have a good night
Patrick

Chorbu, 2023-03-21:
Hello Jan,

The BandPass filter is REACTEL Fcenter 2150 MHz  réf SN86-1 (surely from an old SNA i supposed).

Here are the test comparaison with an ARINST VNA (without calibration made).
You can see behind the stop band filter for 2m fréquency

[image]
[image]

Bests Regards
Patrick

Chorbu, 2023-03-21:
Jan,
Here 2150 Bp Filter with the LibreVNA spectrum analiser with tracking

[image]

And Here the 2m Stop band filter with my old HP8568B analyser with HP8444A tracking generator

[image]

jankae, 2023-03-22:
Okay, did you check it with 10 Hz IF bandwidth as well? Just to rule out the settling time issue I mentioned.

Chorbu, 2023-03-22:
Hi Jan,

I've just made the test with 10 Hz IF BW and it seems good with the 2150 MHz bandPass filter.

[image]

I will try with the 2m stop band filter.

Chorbu, 2023-03-22:
Same with 40 Hz IF BW 

[image]

Chorbu, 2023-03-22:
So it works well under30 Hz IF BW, so i think you can close this PR.

Many thanks, have a good evening.
Patrick

jankae, 2023-03-22:
Okay, thanks for all the additional measurements. Some background information on the "problem", because you might notice that other VNAs can work at higher IF bandwidths: what matters is the overall time after which the stimulus signal is stable. If there is a sufficient delay before sampling the ADCs, you can also take measurements at much higher IF bandwidths. Some VNAs also have a special mode to introduce additional delay there. The LibreVNA currently has no option for any delay (apart from the fixed delay due to the locking of the PLLs) and it is also optimized for fast sweep speed.

Selecting a lower IF bandwidth has a similar effect because more ADC samples are needed (which takes more time). As long as most of the samples are taken when the DUT has fully settled, you will get valid results. Otherwise you could end up with very noisy data - sometimes only in some frequency bands.

I'll go ahead and close this now.

closed 2023-03-22 by jankae