jankae/LibreVNA #204: Win7SP1 x64 drivers problem

DickyWh, 2023-03-24:
### LibreVNA Version

LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 7 Version 6.1 (Build 7601: SP1)
CPU Arch: x86_64

### Steps to reproduce

- LibreVNA app is lauched
- LibreVNA isn't connected throught app, because drivers isn't installed

### Expected behavior

Should work )

### Extra information & Setup and Calibration files

When i connect LibreVNA to another PC with Win10 - all (drivers and soft) installed automatically and work without any problem
How i can install property drivers to Win7SP1 x64?

[image]

jankae, 2023-03-24:
Hm, looks like Windows 7 doesn't load the driver automatically out of the box: https://community.silabs.com/s/article/manually-install-winusb-driver-on-windows-7

If that doesn't help, please try to install the WinUSB driver manually for the LibreVNA. You can use Zadig to do that: https://zadig.akeo.ie/

But you will be mostly on your own with the troubleshooting. I don't have any Win7 available here, sorry.

DickyWh, 2023-03-24:
Thank you/
Zadig resolve my problem!

closed 2023-03-24 by jankae