jankae/LibreVNA #206: Inaccuracies in transfer characteristics graphs at very small Primary Y axis divisions.

YuryW, 2023-03-29:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-7ebb1ea0c
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

### Steps to reproduce

LibreVNA application is running, LibreVNA device is not connected.
At very small divisions of the primary Y-axis, their display occurs with errors.
[image]
[image]

### Expected behavior

I expect correct display of small divisions of the primary Y-axis up to a step of 0.01dB (0.015dB).

### Extra information & Setup and Calibration files

No.

jankae, 2023-04-05:
Huh, this was a floating point issue. 0.01 is actually stored as 0.009999...
This would normally not be a problem, the string conversion function does round the value. But my own function does some preprocessing to determine the prefix (if any should be used) and the number of displayed digits. In this case, the number of displayed digits (or precision) was determined to be 2 (leading zero does not count). 0.009999.. has just zeros for these two displayed digits, which resulted in abbreviating the whole thing as 0.
I have changed the calculation and take into account the remaining digits now as well to determine whether the value will be rounded up. Now the correct values show up:
[image]

jankae, 2023-04-11:
Closing, as I believe this to be solved. Please reopen if you find any further issues related to this.

closed 2023-04-11 by jankae

YuryW, 2023-04-11:
Hi and thanks, Jankae.
Unfortunately, I was temporarily unable to check the new edition, since I mainly work on W7_64, and it does not support Qt6.
Best regards.