jankae/LibreVNA #207: SCPI: unable to open file nor save file

Skippy-the-bush-kangaroo, 2023-04-19:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-1aaba2172
OS: Ubuntu 22.04.1 LTS
CPU Arch: x86_64

### Steps to reproduce

### Saving

I'm unable to save a calibration file or a load a calibration file using SCPI. 
There is no file to be found in the project after a save action and no proper debug message to cling onto. 

```
:VNA:CAL:SAVE TEST
ERROR
```

### Loading

A load action also fails. I've relocated and renamed the DUMMY.CALLKIT file to TEST.calkit. The debug message seem to imply the JSON format is wrong.

```
:VNA:CAL:LOAD? TEST
False

```
 20608.672: [debug] Attempting to open calibration from file "TEST"
 20608.672: [debug] Associated calibration kit expected in "TEST.calkit"
 20608.672: [debug] Opening calkit to file "TEST.calkit"
 20608.673: [debug] new JSON format detected
 20608.676: [warning] "Unable to open file: TEST" 

Maybe I could simply rename the .setup file to a .calkit file? The result would seem obvious to a LibreVNA developer but this confirmed to me this behaviour is a bug, because the setup file is clearly a legacy format.

21384.243: [debug] Attempting to open calibration from file "TEST1"
21384.243: [debug] Associated calibration kit expected in "TEST1.calkit"
21384.243: [debug] Opening calkit to file "TEST1.calkit"
21384.244: [debug] Legacy format detected
21384.253: [warning] "Unable to open file: TEST1"

### Other observations: 

1 The file extension seems to be expected as .calkit while I was assuming like the GUI does, it would be saved as a .config file.
2 In one of the folders there is a file called DUMMY.CALKIT, however the LibreVNA-GUI application (started form cli) seems to want to search a file with a lowercase extension. 

20608.672: [debug] Attempting to open calibration from file "TEST"
20608.672: [debug] Associated calibration kit expected in "TEST.calkit"

3 The manual states: The file path must be relative or absolute. However this seems not possible because the manual also states the filename will be converted to uppercase and Linux is a case sensitive OS. 

 19571.351: [debug] Attempting to open calibration from file "/HOME/USER/LIBREVNA/SOFTWARE/INTEGRATIONTESTS/DUMMY.CALKIT"
 19571.351: [debug] Associated calibration kit expected in "/HOME/USER/LIBREVNA/SOFTWARE/INTEGRATIONTESTS/DUMMY.calkit"
 19571.351: [debug] Parsing of calibration kit failed while opening calibration file:  Unable to open file  (ignore for calibration format >= 3)
 19571.354: [warning] "Unable to open file: /HOME/USER/LIBREVNA/SOFTWARE/INTEGRATIONTESTS/DUMMY.CALKIT"

### Expected behavior

Saving and loading of the config file.

### Extra information & Setup and Calibration files

..

jankae, 2023-04-19:
Hi,
you are raising a couple of valid points (e.g. upper case conversion in a case-sensitive file system). Overall, the calibration save and load command is not my favorite of all the SCPI commands...

I'll try to address all points in the order you wrote them down:

> I'm unable to save a calibration file or a load a calibration file using SCPI.
There is no file to be found in the project after a save action and no proper debug message to cling onto.

I can save a calibration file just fine (no error response from the SCPI server and the file is created on the disk). The only way I can get it to fail is if I attempt to save the file **before** loading (or creating) a calibration in the GUI. And that is intentional - if there is nothing to be saved, it should respond with 'ERROR'.

> A load action also fails. I've relocated and renamed the DUMMY.CALLKIT file to TEST.calkit. The debug message seem to imply the JSON format is wrong.

> Maybe I could simply rename the .setup file to a .calkit file? The result would seem obvious to a LibreVNA developer but this confirmed to me this behaviour is a bug, because the setup file is clearly a legacy format.

These two statements lead me to believe that the different files can be a bit confusing. There are three main different file extensions used with the GUI (along with other file extensions that are not relevant here):
```.setup``` Contains the GUI setup. This consists of the graphs, traces, markers, configured sweep settings,... Notably it does **not** contain the calibration. I made that decision because loading an old calibration along a setup file will probably result in user error and inaccurate measurements - you should always calibrate directly before the measurement. This file extension is not really related to this issue.
```.calkit``` Contains the definitions of the calibration kit standards. There are several legacy file formats for that, some with just the values as ASCII in a particular order, the later (and current) formats use json. Unless you want to store the definitions of multiple calibration kits, you normally don't need to handle these files.
```.cal``` Contains a calibration. This includes the measurements taken during calibration and the type of applied calibration. Similar to the .calkit file, there are legacy formats but the newer formats are json as well. Old formats expect a .calkit file with the same filename as well, newer formats have the calibration kit information embedded in the .cal file.

I hope this overview helps clear things up. In particular, renaming a .setup file to a .calkit file will not help. At best you get a parsing error because it does not contain the expected json fields.

Also, don't try to load .calkit files as a calibration. That won't work as well. Unless you have older files, you likely don't need to bother with .calkit files at all and just save/load .cal files.

> 1 The file extension seems to be expected as .calkit while I was assuming like the GUI does, it would be saved as a .config file.

The ```:VNA:CAL:LOAD?``` command requires the file to be specified with the file ending. There is a further quirk to work around the upper case convention of SCPI commands: the file ending is converted to lower case internally.
The ```:VNA:CAL:SAVE``` command adds the file ending automatically if not specified. Yeah, not really intuitive... All I am doing in the background is calling the normal file loading/saving functions and that is how they are set up because it works well with the FileChooserDialog.

So, this should work:
1. Start the GUI, load any calibration file
2. Save the file via SCPI:
```
:VNA:CAL:SAVE TEST
```
or
```
:VNA:CAL:SAVE TEST.CAL
```
Both should result in a file ```TEST.cal``` in the same folder as your GUI executable.

3. Load the saved calibration again:
```
:VNA:CAL:LOAD? TEST.CAL
TRUE
```

Skippy-the-bush-kangaroo, 2023-04-19:
Thank you for your quick and elaborate reply.  

I was not realizing these where different files altogether. I deducted from an earlier issue posted by another user this was indeed the same file with a renamed extension.  

There was, like you suspected,  no calibration performed, as I'm implementing these SCPI commands into a Python application, one by one. 
I will try to perform a calibration as soon as possible to confirm this calibration file loading issue as soon as possible:  But: ... My LibreVNA has no firmware,  I don't seem to be able to flash a new one from the releases. Strangely enough.
I probably will create a new issue in the coming days if this issue persists.

In the meantime:  Might it be a good idea to include a dummy calibration file in de sources for testing purposes?

Thank you so much with your kind assistance.

jankae, 2023-04-19:
> But: ... My LibreVNA has no firmware

How did that happen? Did it arrive that way? Or did something happen during an update? What to the LEDs on the LibreVNA do when you power it up?

Skippy-the-bush-kangaroo, 2023-04-20:
I really don't know how this happened. I could swear I received the device programmed as it was. The device connects to LibreVNA-GUI but then I had the power LED active on (green) and the one next to it: Booting was flashing two times in red. 

**Successful update:** 
I went to a colleague of mine and he managed to update the device just now to version 1.4.0. One minor strange thing he noticed: He has Windows 11 while the GUI indicates Win10.  His OS is natively installed. 

```
LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 10 Version 2009
CPU Arch: x86_64
```

```
Edition    Windows 11 Pro
Version    21H2
Installed on    ‎23/‎05/‎2022
OS build    22000.1696
Experience    Windows Feature Experience Pack 1000.22000.1696.0
```

**Previously failed**
Ubuntu 22.04 as a virtual machine:  I tried every firmware version from 1.4.1 to 1.2.1. Every time I've got this complaint: 

Evaluating file...
Invalid magic header constant

LibreVNA Version (64 bit): 1.5.0-alpha.2-1aaba2172
OS: Ubuntu 22.04.1 LTS
CPU Arch: x86_64

**Windows 11 VM attempt: Fail**
Windows 11 in a virtual machine where I got a timeout when I tried to update the device.  Response timed out, directly after the verification of the firmware.

```
LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 10 Version 2009
CPU Arch: x86_64
```

**Again a windows 11 VM attempt : Success**
Another day, another try. "Captain's Log, Stardate 44566.7:  I've managed to update via this VM to firmware 1.4.1

The only 2 factors I can think of that changed compared to yesterdays attempt are: 

- My colleague just performed the successful update to 1.4
- I've now got another PSU which can deliver 10 Watt, yesterday, I used a phone charger, although I think it was sufficient. 

```
LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Windows 10 Version 2009
CPU Arch: x86_64
```

**Thank You**
So the issue is settled then, I've got a working LibreVNA and I keep a close look on possible future ocurences.
Thank you so much for the assistance.

Skippy-the-bush-kangaroo, 2023-04-20:
It appears the SCPI command, works now:

I've downgraded the application to:

```
`LibreVNA Version (64 bit): 1.4.0-7f6b78992
OS: Ubuntu 22.04.1 LTS
CPU Arch: x86_64`
```
I'm now opening "TESTFILE.cal" file, with the extension. 

Then, I've copied the content of file "Port 1 1.00M-6.00G 501pt.calkit"  to TESTFILE.calkit and I did the same for the .cal file. 

**This is the result:** 

```
4285.569: [debug] Attempting to open calibration from file "TESTFILE.cal"
  4285.569: [debug] Clearing all calibration measurements...
  4285.569: [debug] Deleted "Port 1 Open" measurement
  4285.569: [debug] Deleted "Port 1 Short" measurement
  4285.569: [debug] Deleted "Port 1 Load" measurement
  4285.569: [debug] Deleted "Port 2 Open" measurement
  4285.569: [debug] Deleted "Port 2 Short" measurement
  4285.569: [debug] Deleted "Port 2 Load" measurement
  4285.569: [debug] Deleted "Isolation" measurement
  4285.569: [debug] Deleted "Through" measurement
  4285.569: [debug] Deleted "Line" measurement
  4285.569: [debug] Error terms reset
  4285.569: [debug] Associated calibration kit expected in "TESTFILE.calkit"
  4285.569: [debug] Opening calkit to file "TESTFILE.calkit"
  4285.570: [debug] JSON format detected
  4285.597: [debug] Clearing all calibration measurements...
  4285.597: [debug] Deleted "Port 1 Open" measurement
  4285.597: [debug] Deleted "Port 1 Short" measurement
  4285.597: [debug] Deleted "Port 1 Load" measurement
  4285.597: [debug] Deleted "Port 2 Open" measurement
  4285.597: [debug] Deleted "Port 2 Short" measurement
  4285.597: [debug] Deleted "Port 2 Load" measurement
  4285.597: [debug] Deleted "Isolation" measurement
  4285.597: [debug] Deleted "Through" measurement
  4285.597: [debug] Deleted "Line" measurement
  4285.597: [debug] Error terms reset
  4285.605: [debug] Checking if "Port 1" calibration is possible...
  4285.605: [debug] ...calibration possible
  4285.605: [debug] Checking if "Port 1" calibration is possible...
  4285.605: [debug] ...calibration possible
  4285.605: [debug] Constructing error terms for "Port 1" calibration
  4285.612: [debug] Checking if "Port 1" calibration is possible...
  4285.612: [debug] ...calibration possible
  4285.612: [debug] Checking if "Port 1" calibration is possible...
  4285.612: [debug] ...calibration possible
  4285.612: [debug] Constructing error terms for "Port 1" calibration

```

closed 2023-04-21 by Skippy-the-bush-kangaroo