jankae/LibreVNA #210: [Proposal] PC_Application: Replacing QMake with CMake

crvux, 2023-04-28:
### Summary

Modern Qt6 uses CMake as its main build system. QMake is deprecated
https://www.qt.io/blog/2018/10/29/deprecation-of-qbs

How about switching to CMake?

### Paths to implement it

I can write new CMakeList and test build on Linux. I need help testing the build process on Windows and Mac
Also, there is need some сhanges in CI\CD. I think they will be minor and we will deal with them

### Extra information

Some CMake example from Qt
https://doc.qt.io/qt-6/cmake-get-started.html

jankae, 2023-04-28:
From your link:

> Support for qmake will continue unaffected

and

> For Qt users, qmake is currently the most widely used build system. CMake is a clear second and growing in popularity.

Doesn't sound like qmake is deprecated. I also don't see the advantage of switching the build system. It will just generate more work without resulting in more features. Or am I missing anything obvious here?

crvux, 2023-04-30:
Sorry for misleading. I mistakenly assumed that if a Qt6 now uses CMake for build, then QMake is deprecated. You are right, QMake is continues to be supported, but patches are limited to fixes only (https://code.qt.io/cgit/qt/qtbase.git/stats/qmake?h=dev&period=q&ofs=-1).

Below are some opinions about migration from QMake to CMake
https://www.reddit.com/r/QtFramework/comments/zldcf5/what_is_the_future_of_qmake_pro_files/
https://pvs-studio.com/en/blog/posts/cpp/0967/

Key points when using CMake
- CMake is more popular and potentially attract more contributors
- CMake has very good support in modern IDE like Visual Studio Code and CLion
- If you plan to extend the project with third-party libraries, then integrating this libraries will be easier
- Cross-compilation potentially can be easier too

IMHO that if there are big plans for the development of the project, it makes sense to move to CMake

crvux, 2023-04-30:
However, there are still widely used projects that are built with legacy tools (GNU autotools for example) and this does not prevent their development, although this process is less convenient

jankae, 2023-04-30:
Thanks for making me aware that Qt can use CMake as the build system, I though it was limited to qmake only.

But I don't really see the advantage of CMake for this project:
- I am not very experienced with CMake
- The latest QtCreator version still defaults to qmake when creating a new project
- Integrating other libraries ([json](https://github.com/nlohmann/json) and [math parser](https://github.com/beltoforion/muparserx)) was really easy with qmake as well
- There already exists working CI/CD for qmake
- Switching to CMake would not result in any immediate improvements (apart from possibly making it easier for other people to contribute - but I think anyone who has even a little bit of experience with Qt will be familiar with qmake as well)

closed 2023-04-30 by jankae

mv2050arshad, 2024-05-10:
> ### Summary
> Modern Qt6 uses CMake as its main build system. QMake is deprecated https://www.qt.io/blog/2018/10/29/deprecation-of-qbs
>
> How about switching to CMake?
>
> ### Paths to implement it
> I can [...]

Hi, i have same problem. Could you converted it to cmake ?

jankae, 2024-05-10:
Hi, I think my arguments on this topic are unchanged. If in many years this project switches to Qt7 it likely make sense to switch to cmake at the same time. But until then, I can not see any benefit of doing so:

- It will be take some work to migrate the CI/CD for all platforms
- For users that only use the binaries (i.e. do not compile themselves) there is no difference/benefit at all
- For users (like you) that want to compile with cmake, it may be beneficial, but:
  - There are build instructions using qmake in this repo
  - These instructions are not that complicated

In conclusion: No, I will not convert it to cmake at this time. But I am happy to help with any problems you encounter while building this project with qmake

mv2050arshad, 2024-05-10:
> Hi, I think my arguments on this topic are unchanged. If in many years this project switches to Qt7 it likely make sense to switch to cmake at the same time. But until then, I can not see any [...]

Thanks for your fast answering , you are right. I dont have problem with build in qmake.