jankae/LibreVNA #213: Mistake in calculation of quality factor

AlaiziyZukerman, 2023-07-24:
### LibreVNA Version

os Windows, last release

### Steps to reproduce

- In GUI is calculation of reactance and resistance, this parameters is correct. There is quality factor calculation, and this calculation is mistaked becaus quality factor = | reactance / resistance |. Manualy calculation is not equal result in GUI.

### Expected behavior

quality factor = | reactance / resistance |

### Extra information & Setup and Calibration files

- Please, add |Z| (complex resistance module) numeric and graphical in GUI.

AlaiziyZukerman, 2023-07-24:
yes, I mean magnitude of the complex impedance. measure it in Ohm like resistance or reactance.

jankae, 2023-07-24:
Did you see these options?
[image]
I think they should cover that (unless I am still not understanding what you mean)

AlaiziyZukerman, 2023-07-24:
[image]
reactance and resistance is a usable options, but I need |Z| like on image.

jankae, 2023-07-24:
There is no option to plot the magnitude of the impedance on a graph. I didn't really think anyone would need that, since it loses the vector information. These are your options right now:

- if you need that value only at one (or a few points):
  - Use a cursor and configure it to show impedance (still split into real and imaginary resistance)
  - Configure the mouse-over text on the smith chart to show impedance (right-click->Cursor Format->Impedance)
- if you need that value for all points of the trace:
  1. Create a custom math expression for the trace which converts the S parameter data to the magnitude of the impedance:
  [image]
  2. Select "real" as the Y-axis type:
  [image]
  This will result in the graph simply showing the result of your custom math expression
  3. You can now see the magnitude of the impedance across frequency:
  [image]
  4. Keep in mind that custom math expressions are a powerful tool but can also result in confusing graphs/data. Since you can enter any formula you want, the GUI doesn't have a good idea anymore what the data actually represents. It just pretends that it is still S parameter data. This means you can for example add markers to the trace but the marker values might not be meaningful (depending on what exactly you used as the custom expression)

closed 2023-07-25 by AlaiziyZukerman