jankae/LibreVNA #215: Spikes at 31MHz and 62MHz

Dmitrii1978, 2023-09-13:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-
OS: Windows 11 Version 22H2
CPU Arch: x86_64

### Steps to reproduce

Spikes at 31MHz and 62MHz and higher multiples frequencies.
[image]

### Expected behavior

Is it possible to decrease level of these spikes?

### Extra information & Setup and Calibration files

The frequency range is set to 1MHz..101MHz , 4501 points.

jankae, 2023-09-13:
These spikes are caused by LO feedthrough (and mixing products of the two LOs). I don't think it is possible to remove them without drastic changes to the hardware, sorry.
You have the option to shift them to slightly different frequencies by changing the LO frequencies: Device->Configuration
[image]

This could be useful if you need to look at the actual signal right at these frequencies. But I would recommend changing the LO only by 1-2 MHz, otherwise the hardware filters may attenuate the signal too much (increased noise).

YuryW, 2023-09-20:
> ### LibreVNA Version
> LibreVNA Version (64 bit): 1.5.0-alpha.2- OS: Windows 11 Version 22H2 CPU Arch: x86_64
> 
> ### Steps to reproduce
> Spikes at 31MHz and 62MHz and higher multiples frequencies. [image]
> 
> ### Expected behavior
> Is it possible to decrease level of these spikes?
> 
> ### Extra information & Setup and Calibration files
> The frequency range is set to 1MHz..101MHz , 4501 points, IF BW 1.00kHz

Hi, Dmitrii1978. I have a slightly different picture in this frequency range.
[image]

It is possible that I have a different LibreVNA device (six-layer board and added LO harmonic filters).
https://github.com/jankae/LibreVNA/issues/161#issuecomment-1372602939

Perhaps such filters already produce better results than yours.
Often bursts can be significantly reduced by changing one frequency limit, for example, the lower one at 250 kHz. This is my result.
[image]

Even changing the number of scanning points by one helps to shift the parasitic bursts in frequency.
[image]
[image]

But how many parasitic bursts there are even with harmonic filters of the first local oscillator, if you use scanning boundaries without a shift of 250 kHz, that is, as in the case you presented.
[image]

And of course, reducing scanning points can completely eliminate parasitic bursts.
[image]

closed 2023-10-23 by jankae