jankae/LibreVNA #224: The ConnectToFirstDevice functionality does not respect the restricted device list created by the command line option

SultanAbughazal, 2023-10-31:
### LibreVNA Version

LibreVNA Version (32 bit): 1.4.0-6a27d3944
OS: Raspbian GNU/Linux 11 (bullseye)
CPU Arch: arm64

### Steps to reproduce

- Connect 2 or more LibreVNA device to your computer.
- Launch the LibreVNA app with the `--device` option using the serial number of one of the LibreVNA devices.
- Observe the debug logs; specifically the serial number of the device it connected to.
- Repeat for all other LibreVNA devices connected to the computer.
- Observe that the app connects to the same LibreVNA device regardless of the serial number specified in the command line.

### Expected behavior

- The app uses the restricted device option in the devices drop down menu. (works)
- The app uses the restricted device option when connecting to the first device. (does not work)

### Extra information & Setup and Calibration files

Fresh install, no specific set up or calibration.

jankae, 2023-10-31:
Can you please try with the latest commit as well? This should be fixed since cde56429

jankae, 2023-10-31:
nevermind, still not fixed. Working on it now

jankae, 2023-10-31:
...and now it should really be fixed: 4c0d18b
Tests before the fix confirmed the behavior you reported. After the fix, the GUI only connects to the specified device.

closed 2023-11-03 by SultanAbughazal