jankae/LibreVNA #225: Inability to view S11, S22, S33 of s3p file on Smith chart.

YuryW, 2023-10-31:
### LibreVNA Version

LibreVNA Version (64 bit): 1.5.0-alpha.2-99c6d77ad
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
CPU Arch: x86_64

### Steps to reproduce

Launch the LibreVNA application. We do not connect the LIbreVNA device. Download the s3p file (TCP-2-152-75X+) from the original source (M[ini-Circuits](https://www.minicircuits.com/WebStore/dashboard.html?model=TCP-2-152-75X%2B)). Open it in LibreVNA software. We right-click on any Smith chart and do not see the opportunity to look at S11, S22, S33 of the downloaded s3p file.
[image]

### Expected behavior

Ability to view s3p files and in Smith charts for the LibreVNA program

### Extra information & Setup and Calibration files

[TCP-2-152-75X+_S3P.zip](https://github.com/jankae/LibreVNA/files/13219360/TCP-2-152-75X%2B_S3P.zip)

jankae, 2023-10-31:
This particular touchstone file is saved with a reference impedance of 75 Ohm instead of the usual 50 Ohm. As a result, the GUI is unable to show it by default, because the smith chart (on startup) is configured for 50 Ohm reference impedance.

You have two options here (actually only one, but two ways to achieve the same result):
1. change the reference of the smith chart to 75 Ohm:
[image]
You need to remove the default S11 or S22 trace first, otherwise the trace causes the reference impedance to be changed back to 50 Ohm immediately.
Afterwards, you should be able to add the traces from the touchstone file:
[image]

2. Just drag-and-drop the trace from the touchstone file onto the smithchart (without changing the reference impedance first). This is a shortcut and will (after a warning) remove the previous enabled S11/S22 trace as well as set the reference impedance of the smith chart to the correct value.

YuryW, 2023-10-31:
Thanks Jan. 
Both options worked for me.

closed 2023-10-31 by YuryW